Screenly/Anthias #248: Live Video Stream FAIL

fabltd, 2015-01-06:
Hello

I am having a strange problem with Screenly. I have a live stream server with the URL http://192.168.1.10:8099/test.mp4 if I shh into the PI and run omxplayer followed by the URL the stream plays with no issue. If i load the URL into Screenly I get "Server Error: Could not retrieve file. Check the asset URL" how do i debug to see why its failing ? I understood screenly should just pass the URL to OMXplayer. 

Thanks

vpetersson, 2015-01-06:
We don't support streaming video in Screenly at the moment. This support was added in `omxplayer` in relatively recent time and we haven't added support for this. 

It is likely not too difficult to add support for, but would require some work.

fabltd, 2015-01-06:
Is it supported in PRO ?

vpetersson, 2015-01-06:
No, not at the moment.

fabltd, 2015-01-06:
Oh that's a shame. I have an install that this would be ideal for. How difficult is it to pass the URL direct to OMXPlayer ?

vpetersson, 2015-01-06:
It's probably not too difficult to add this, and we welcome a pull request for this for OSE as we have no immediate plans for adding support for this.

In terms of Pro, we're demand driven. Hence, if we see enough demand for it (which we haven't at this point), we'll likely add support for it.

fabltd, 2015-01-07:
Can you suggest any one who might be interested in making the changes?

vpetersson, 2015-01-07:
Not off the top of my head, but it shouldn't be too difficult. I'm sure you can find someone off of oDesk or similar.

xbmcnut, 2015-03-03:
I have this working with two different streams; rtsp://video2.earthcam.com/fecnetwork/4559.flv and 
rtsp://85.255.175.244:554/h264 however, screenly will not save the video duration settings meaning that whatever video starts playing first, it will not stop. The asset is registered as a video and if changed to a webpage, of course, it will not render. If the video duration would stick, this would be an excellent option for embedding live stream into a screenly playlist. Version installed was 2015-02-25.

vpetersson, 2015-09-23:
Duplicate of #90.

closed 2015-09-23 by vpetersson