Screenly/Anthias #249: Server Error: Could not retrieve file. Check the asset URL.

nalandry, 2015-01-15:
I received this error when trying to add a public Google Calendar. URL is https://www.google.com/calendar/embed?src=8p0c18obrslfdl18s3eii8io1o%40group.calendar.google.com&ctz=America/New_York

richard-to, 2015-01-15:
@nalandry @vpetersson 

I was able to reproduce this bug. I'll take a look at it later today.

vpetersson, 2015-01-15:
@richard-to I thought this was resolved some time ago, but perhaps it resurfaced. Here are two similar tickets from the past:

 * #188
 * #124

richard-to, 2015-01-16:
Turns out a HEAD request to the Google Calendar URLs return 404. I have no idea why. If you do a GET request, then you get 200.

richard-to, 2015-01-16:
One possible solution is to try the HEAD request first and if that fails, fall back to a GET request. Or alternatively just do a GET request at the expense at some extra bandwidth.

vpetersson, 2015-01-16:
@richard-to Very good catch. I got a 404 when i tried with Curl, but i thought it was a User Agent thing. 

Adding a fallback to pull down the full page in case we get a 404 could be a reasonable approach. Alternatively, we da custom handling for `www.google.com`.

@aljungberg what do you think?

over64, 2016-02-02:
according to https://github.com/wireload/screenly-ose/pull/335

vpetersson, 2016-04-27:
This has been landed and now works in master.

closed 2016-04-27 by vpetersson