Screenly/Anthias #250: Browser found dead

stevenscg, 2015-01-15:
I wanted to provide this issure report for completeness, though I am currently exploring the possibility of some bad 64GB Transcend micro SD cards (P/N: TS64GUSDU1E).  I am happy to help troubleshoot as needed, but will likely try a Samsung or other well supported SD card.

When running with the above cards and only 2-3 webpages loaded for assets, the screen will fail to render.

Sometimes this appears to happen even immediately after reboot.  When screenly content is visible after a reboot, it usually stops after some time (minutes to hours).  In this condition, the screen is black with occasional flashes of white or gray and the log files show "browser found dead, restarting".

This issue does not seem to occur when using the 8GB cards from the CanaKit Raspberry Pi B+ Complete Starter Kit.

These PIs run only the Screenly OSE process on Raspbian (`2014-09-09-wheezy-raspbian`).  We have tried `2014-12-24-wheezy-raspbian`, but that seemed even less stable.

Hardware:
Raspberry Pi B+ (2 separate devices)
64GB Transcend micro SD cards (P/N: TS64GUSDU1E)
Network: Ethernet

```
pi@raspberrypi ~ $ tail -f /tmp/screenly*.log
==> /tmp/screenly_viewer.log <==
Thu, 15 Jan 2015 09:53:29 browser found dead, restarting
Thu, 15 Jan 2015 09:53:29 Loading browser...
Thu, 15 Jan 2015 09:53:33 Loading browser...
Thu, 15 Jan 2015 09:53:33 Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 5158.
Thu, 15 Jan 2015 09:53:37 Generating asset-list...
Thu, 15 Jan 2015 09:53:38 Showing asset Librato Dashboard - HQ (webpage)
Thu, 15 Jan 2015 09:53:38 browser found dead, restarting
Thu, 15 Jan 2015 09:53:38 Loading browser...
Thu, 15 Jan 2015 09:53:41 Loading browser...
Thu, 15 Jan 2015 09:53:41 Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 5191.

==> /tmp/screenly_xloader.log <==
EVENT [5158] VARIABLE_SET status_format str '<b>@[@TITLE]@</b> - @[@uri]@ - <span foreground="#bbb">@NAME</span>'
EVENT [5158] VARIA... (981 more, please see e.stdout)

  STDERR:

(uzbl-core:5158): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-label-select-on-focus after class was initialised

(uzbl-core:5158): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-scrolled-window-placement after class was initialised

Cleaning up...

==> /tmp/screenly_viewer.log <==
Thu, 15 Jan 2015 09:53:46 Generating asset-list...
Thu, 15 Jan 2015 09:53:47 Showing asset Librato Dashboard - HQ (webpage)
Thu, 15 Jan 2015 09:53:47 browser found dead, restarting
Thu, 15 Jan 2015 09:53:47 Loading browser...

==> /tmp/screenly_xloader.log <==
Traceback (most recent call last):
  File "/home/pi/screenly/viewer.py", line 364, in <module>
    main()
  File "/home/pi/screenly/viewer.py", line 360, in main
    asset_loop(scheduler)
  File "/home/pi/screenly/viewer.py", line 300, in asset_loop
    browser_url(uri)
  File "/home/pi/screenly/viewer.py", line 175, in browser_url
    browser_send('uri ' + current_browser_url, cb=cb)
  File "/home/pi/screenly/viewer.py", line 160, in browser_send
    load_browser()
  File "/home/pi/screenly/viewer.py", line 131, in load_browser
    if browser:
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 334, in __len__
    return len(str(self))
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 370, in __str__
    else: return unicode(self).encode(self.call_args["encoding"])
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 373, in __unicode__
    if self.process and self.stdout:
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 316, in stdout
    self.wait()
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
    self._handle_exit_code(self.process.wait())
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
    self.process.stderr
sh.SignalException_11: 

  RAN: '/usr/bin/uzbl-browser --config=- --uri=file:///tmp/screenly_html/black_page.html --print-events'

  STDOUT:
EVENT [5191] INSTANCE_START 5191
EVENT [5191] BUILTINS 'back forward scroll reload reload_ign_cache stop zoom_in zoom_out toggle_zoom_type uri js script toggle_status spawn sync_spawn sync_spawn_exec sh sync_sh exit search search_reverse search_clear dehilight set toggle dump_config dump_config_as_events chain print event request menu_add menu_link_add menu_image_add menu_editable_add menu_separator menu_link_separator menu_image_separator menu_editable_separator menu_remove menu_link_remove menu_image_remove menu_editable_remove hardcopy include show_inspector add_cookie delete_cookie clear_cookies download '
EVENT [5191] VARIABLE_SET status_format str '<b>@[@TITLE]@</b> - @[@uri]@ - <span foreground="#bbb">@NAME</span>'
EVENT [5191] VARIA... (981 more, please see e.stdout)

  STDERR:

(uzbl-core:5191): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-label-select-on-focus after class was initialised

(uzbl-core:5191): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-scrolled-window-placement after class was initialised

Cleaning up...
```

vpetersson, 2015-01-15:
Hi @stevenscg. Thanks for the detailed report. However, I think this might be a Raspberry Pi issue. Apparently the Raspberry Pi struggles with some 64GB drives. Hence, I think it may be related to that (and in turn causes Screenly to act up). This would also explain why it works just fine with the 8GB card.

stevenscg, 2015-01-15:
Great to know about the 64GB drives and the Pi in general!

I actually just ordered some of the SanDisk Extreme 32GB UHS-I/U3 Micro SDHC cards to try out.  I will post back here and close the issue if everything clears up.

vpetersson, 2015-01-15:
@stevenscg Thanks. The more information we can collect about various cards, the better.

stevenscg, 2015-01-18:
The new Samsung microSD cards are in along with a brand new B+ pi. I am trying to test, but now I cannot seem to get screenly to load any assets.

This is the latest raspbian (2014-12-24) + screenly OSE install (via curl-bash).  I seem to recall having some trouble with this release of raspbian vs the previous version (september, if I recall) on my other B+ models.

I'll try to load up the older Raspbian this afternoon, but @vpetersson, have you experienced a fresh install that only starts up with these screenly processes running? 

```
pi@pi3 ~ $ ps ax | grep screen
 2820 ?        S      0:02 python /home/pi/screenly/server.py
 2835 ?        Sl     0:08 /usr/bin/python /home/pi/screenly/server.py
 2850 pts/0    S+     0:00 grep --color=auto screen
``` 

There is currently 1 webpage being scheduled via url, but I don't see any attempts to load it either:
[image]

vpetersson, 2015-01-18:
@stevenscg Hmm, can you check if `viewer.py` and `xloader.sh` are running? 

Also, I haven't tested the install script on a recent version of Raspbian as the primary delivery mechanism these days are using our disk images.

stevenscg, 2015-01-19:
I ran out of time to check those, but it might be something to watch out for when you upgrade Raspbian for your pre-built images.

I am actually running your 11/03 image right now on the SanDisk Extreme 32GB UHS-I/U3 Micro SDHC cards and it is very stable several hours in and video playback.

I'll close this issue as it looks like general instability of the Pi with the 64GB cards.

closed 2015-01-19 by stevenscg

vpetersson, 2015-01-19:
Thanks for the update @stevenscg. Will keep an eye open for that. However, we have our own apt-mirror that we use for that precise reason :)