Screenly/Anthias #252: Assets wont show

jmarkwick, 2015-01-26:
I chose to download one of the screenly images from https://www.screenlyapp.com/ose.html.
Screenly does start as normal but i am unable to add assets. The preset images display fine i.e Big Buck Bunny Trailer and i am able to turn these on and off etc. 

When i add a new asset it all looks fine. It just doesnt show on the screen.

Please help

vpetersson, 2015-01-26:
Hmm that's strange. Would you mind posting the logs?

jmarkwick, 2015-01-26:
Sorry, How can i get hold of the logs?

vpetersson, 2015-01-26:
They're stored on the unit. Paste the output of the files matching `/tmp/screenly*`.

jmarkwick, 2015-01-26:
i have the one file.  black_page.html?

vpetersson, 2015-01-26:
No, that's not it.

Also, look in `/var/log/screenly` and `/tmp/screenly` to see if you can see the logs there.

jmarkwick, 2015-01-26:
screenly_xloader.log?
screenly_viewer.log

/var/log/screenly does not exist

vpetersson, 2015-01-26:
yes `screenly_*.log`. Please paste the output of those files.

jmarkwick, 2015-01-26:
Disabling screen power savings...
Launching infinite loop...
Cleaning up...
Traceback (most recent call last):
  File "/home/pi/screenly/viewer.py", line 364, in <module>
    main()
  File "/home/pi/screenly/viewer.py", line 360, in main
    asset_loop(scheduler)
  File "/home/pi/screenly/viewer.py", line 288, in asset_loop
Traceback (most recent call last):
  File "/home/pi/screenly/viewer.py", line 364, in <module>
    main()
  File "/home/pi/screenly/viewer.py", line 360, in main
    asset_loop(scheduler)
  File "/home/pi/screenly/viewer.py", line 288, in asset_loop
    view_image(HOME + LOAD_SCREEN)
  File "/home/pi/screenly/viewer.py", line 181, in view_image
    browser_send('js window.setimg("{0}")'.format(uri), cb=lambda b: 'COMMAND_EXECUT$
  File "/home/pi/screenly/viewer.py", line 160, in browser_send
    load_browser()
  File "/home/pi/screenly/viewer.py", line 131, in load_browser
    if browser:
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 334, in __len__
    return len(str(self))
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 370, in __str__
    else: return unicode(self).encode(self.call_args["encoding"])
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 373, in __unicode__
    if self.process and self.stdout:
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 316, in stdout
    self.wait()
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
    self._handle_exit_code(self.process.wait())
  File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
    self.process.stderr
sh.SignalException_15:

$ --pri$Disabling screen power savings...
Launching infinite loop...

jmarkwick, 2015-01-26:
Mon, 03 Nov 2014 12:47:55 Loading browser...
Mon, 03 Nov 2014 12:47:55 Browser loading http://0.0.0.0:8080/splash_page. Running a$
Mon, 03 Nov 2014 12:49:09 Generating asset-list...
Mon, 03 Nov 2014 12:49:10 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:49:11 current url is file:///tmp/screenly_html/black_page.html
Mon, 03 Nov 2014 12:49:17 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:49:22 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:49:28 Generating asset-list...
Mon, 03 Nov 2014 12:49:28 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:49:33 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:49:39 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:49:45 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:49:50 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:49:56 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:02 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:07 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:13 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:18 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:24 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:29 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:35 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:41 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:46 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:52 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:50:57 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:03 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:09 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:14 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:20 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:26 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:31 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:37 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:43 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:49 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:51:54 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:52:00 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:52:05 Playlist is empty. Sleeping for 5 seconds
Mon, 03 Nov 2014 12:52:11 Playlist is empty. Sleeping for 5 seconds

vpetersson, 2015-01-26:
Ok, looks like the playlist is empty. Make sure the assets show up as 'active'.

jmarkwick, 2015-01-26:
[image]

vpetersson, 2015-01-26:
Hmm, actually look at the timestamp of the logs 'Mon, 03 Nov 2014 [...]' 
Either those logs were really old, or the time is incorrect on the device.

Can you paste the output of the following:

     $ date

and

    $ tail -n 20 screenly_*.log

Oh, and i just noticed in the other logs that something definitely is wrong in `viewer*.log`. Perhaps the SD card is broken. That's a very common reason.

jmarkwick, 2015-01-26:
I just twigged when i posted that last image. The date could be wrong.
I changed the start date to 4 years ago and all works fine. Thanks for your help

vpetersson, 2015-01-26:
Gerat!

closed 2015-01-26 by vpetersson

car-thee-ca, 2019-06-27:
> I just twigged when i posted that last image. The date could be wrong.
> I changed the start date to 4 years ago and all works fine. Thanks for your help

how to change date?

vpetersson, 2019-06-27:
@car-thee-ca please use the [forum](https://forums.screenly.io/c/screenly-ose) for questions.