Screenly/Anthias #253: Multiple instances of the same processes? viewer.py & server.py

abrenoch, 2015-01-29:
I'm not sure if this is a bug or not, but I thought I would run it by you guys just to double check!

I have noticed some performance drops after running screenly for a few hours.. The http/ssh connectivity will become seemingly intermittent, sometimes not connecting at all. I decided to try running htop to inspect what all was eating up the CPU, and have found there are multiple instances of `server.py` and `viewer.py` running. Is this by design or is there something else going on here?

Also I have noticed `uzbl-core --config=- --uri=file:///tmp/screenly_html/black_page.html` occasionally using 60+% on the CPU, and I notice when the process does spike the pi can 100% CPU usage.

Does this all seem normal? I know the pi is low powered so we can only ask so much from it..

vpetersson, 2015-01-29:
> found there are multiple instances of server.py and viewer.py running. Is this by design or is there
> something else going on here?

If that is the case, something is wrong. There should only be one instance each.

> Also I have noticed uzbl-core --config=- --uri=file:///tmp/screenly_html/black_page.html occasionally 
> using 60+% on the CPU, and I notice when the process does spike the pi can 100% CPU usage.

This is expected. That's the web browser, and it usually maxes out the Pi when loading something.

Are you using the official build or a self-installed version?

abrenoch, 2015-01-29:
Thanks for the reply vpete!

Regarding the web browser; That makes sense, however I do not have any 'webpage' assets added.. Would it still be normal that it pops up?

I installed screenly ose via the image provided on the official webpage.. I would be willing to try re-installing it if you think it may help. I haven't made any changes to the OS really except for putting it into portrait mode, as well as a few performance tweaks I attempted after noticing things being a bit slow (Tried new power supply, overclock settings, memory split, and increasing the SWAP size).

Just now actually I came into the office and the display running screenly is still displaying the slides (all 1080p images), however I am not able to access it over the network.. Rebooting seems to fix the problem.

Any advice you may have is greatly appreciated!

vpetersson, 2015-01-29:
> Regarding the web browser; That makes sense, however I do not have any 'webpage' 
> assets added.. Would it still be normal that it pops up in htop?

Yes, the browser is still loaded in the background and used to display some information.

> however I am not able to access it over the network.. Rebooting seems to fix the problem.

Nothing obvious comes to mind. You could perhaps try swapping out the SD card to see if that makes a difference.

abrenoch, 2015-01-29:
I'll keep switching the SD card in mind, I was considering trying that anyway..

I attached an image that shows my htop just after rebooting while things are still running nicely. You can see there are 3 instances of `viewer.py` and 2 of `server.py`. In this shot they are not using too much CPU, but they do seem to slowly grow over time.

I'll keep trying some different ideas, worst come to worst I will just re-install the OS.. I just hooked it up to a slightly more beefy power supply (5.1v/2.1a) so hopefully that will help with the network dropout!

If you have any ideas or debugging you would like me to try please just let me know, I really like the app and would just like to have it running more stable!

vpetersson, 2015-01-29:
Could you post the data from the logs? The relevant ones should be:

 * /tmp/screenly_*.log
 * /var/log/supservisor/*.log

abrenoch, 2015-01-29:
Sure thing!

screenly-access.log: http://paste.debian.net/142948
supervisord.log: http://paste.debian.net/142949
screenly_viewer.log: http://paste.debian.net/142950
screenly_xloader.log: http://paste.debian.net/142951

These where taken at about 15 minutes of uptime, using about 340M of RAM.

Hope you can find something helpful in there, I did see some errors produced in `screenly-acces.log`

Thanks for your help!

vpetersson, 2015-01-29:
There are definitely some errors in the access log, but it's hard to determine what might be causing that. Can you find anything interesting in the system log?

> It may also be worth noting, if I manually kill one of the uzlb-core processes, 
> they all get terminated and the memory is recovered.

Try running `ps faux` to see if those processes are related. Perhaps you killed the parent process.

abrenoch, 2015-01-29:
Here is the output of `ps faux`, I'll check out the system logs shortly here..

`ps faux`:
http://paste.debian.net/142981

vpetersson, 2015-01-30:
Judging by the output of `ps faux` everything looks fine.

abrenoch, 2015-01-30:
Okay, I wont worry about the multiple instances then, maybe its just the way htop is reporting the processes..

Also, I did try turning off the SWAP memory totally.. So far it has been up for 16 hours and everything seems to be working well still! I had originally tried increasing it via recommended 2xMem(if less than 2GB), but after spending some more time reading up on it, it seemed like having it turned off would be the best practice for the raspberrypi considering it runs off an SD card.

So things are actually looking better now! Thanks for the help! I'm eager to dig into what else screenly is capable of!

closed 2015-01-30 by abrenoch

vpetersson, 2015-01-30:
Great. Yeah, so the old 2xMem rule is probably not very well suited for this particular situation.