Screenly/Anthias #256: Raspberry Pi 2 support

craigpelican, 2015-02-02:
Hi after the announcement of the Raspberry Pi 2 today. 
http://www.raspberrypi.org/raspberry-pi-2-on-sale/
Will there be a new downloadable image coming soon ?

vpetersson, 2015-02-02:
We haven't got our hands on the new hardware yet. As such, we don't know if the existing image will work or not. Chances are that it will.

craigpelican, 2015-02-02:
OK. Mine should turn up tomorrow, hopefully.
I'll let you know, unless you beat me to it !

P.S. Sorry am new to Github, couldn't set this as a 'question' rather than an 'issue'.

vpetersson, 2015-02-02:
Ok, let us know how it goes.

Also, there are no 'questions' on Github, as it is largely related to code.

andersekbom, 2015-02-02:
Not sure wether this applies, but:

"Also, any precompiled software will not work at full speed (although supposedly the processor will be able to run it). Still, you'll want to have it recompiled for the new processor! For many people, this isn't a big deal, but if you have a pre-created Pi 1 Model A+B+ card image, just be aware it won't work!"

https://learn.adafruit.com/introducing-the-raspberry-pi-2-model-b

vpetersson, 2015-02-02:
Screenly doesn't include a lot of compiled code (none for OSE actually).

Since the new binaries (compiled for ARMv7 vs ARMv6) will be provided upstream. I don't know exactly how this will play out. Perhaps there will be two different images of Raspbian. In that case, we will have to provide two versions too.

craigpelican, 2015-02-02:
I have just managed to get my hands on a Raspberry Pi 2. It's installing Raspian as I type...
Still pretty new to the Linux field, but I'll see how far I can get using Option 2: (https://www.screenlyapp.com/ose.html) before something catches fire !

vpetersson, 2015-02-02:
Interesting. Let me know how it goes.

Cyberprog, 2015-02-02:
> -----Original Message-----
> 72466995> .

That was quick! I too, am interested in how Screenly performs on the Pi2 :)

Regards,

Alex Threlfall
IT & Communications Manager
POWRED HEATING COMPONENTS LTD.
Tel: 01179415689
Fax: 0117 9414060
alex.threlfall@powred.com
www.powred.com

craigpelican, 2015-02-02:
OK. Well it's up and running, but...
Screenly doesn't run on boot. Im just taken to the desktop, and can't see the mouse cursor.
I can get to the web console via http://<IPADDRESS>:8080 and can upload assets (which does seem pretty nippy in doing so).

But under System Info its not running the video:
[image]

Overview:
[image]

Settings:
[image]

vpetersson, 2015-02-02:
Did you get any errors when running the installer? Perhaps `omxplayer` is yet to be ported.

craigpelican, 2015-02-02:
pi@raspberrypi ~ $ curl -sL https://raw.github.com/wireload/screenly-ose/master/                                                                    misc/install.sh | bash
Installing Screenly OSE
Updating system package database...
Upgrading the system...
(This might take a while.)
Installing dependencies...
grep: /etc/init.d/mathkernel: No such file or directory
Extracting templates from packages: 100%
/run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV in                                                                    vocation.
/run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV in                                                                    vocation.
Downloading Screenly-OSE...
remote: Counting objects: 3027, done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 3027 (delta 2), reused 0 (delta 0)
Receiving objects: 100% (3027/3027), 1.11 MiB | 742 KiB/s, done.
Resolving deltas: 100% (1648/1648), done.
Installing more dependencies...
Adding Screenly to X auto start...
Increasing swap space to 500MB...
Adding Screenly's config-file
Enabling Watchdog...
[ ok ] Stopping watchdog keepalive daemon....
[ ok ] Starting watchdog daemon....
Adding Screenly to autostart (via Supervisord)
Making modifications to X...
Quiet the boot process...
Assuming no errors were encountered, go ahead and restart your computer.
pi@raspberrypi ~ $ sudo reboot

Broadcast message from root@raspberrypi (pts/0) (Mon Feb  2 15:04:46 2015):
The system is going down for reboot NOW!

(Sorry not sure how to show as 'code' rather than plain text)

vpetersson, 2015-02-02:
@craigpelican You simply indent each line with four spaces to make it into code styling. 

However, looking at the log, I don't see anything obvious. 

Can you try `omxplayer --version`?

craigpelican, 2015-02-02:
@vpetersson Thanks for the tip ;)

    pi@raspberrypi ~ $ omxplayer --version
    omxplayer - Commandline multimedia player for the Raspberry Pi
    Build date: Fri, 24 Oct 2014 00:48:49 +0000
    Version   : 98982df [master]
    Repository: https://github.com/popcornmix/omxplayer.git

vpetersson, 2015-02-02:
Hmm, that looks fine. Could you try poking around in /tmp/screenly*.log and /tmp/xloader.log to see if there is anything interesting there?

craigpelican, 2015-02-02:
screenly_viewer.log is empty
and there are no other logs in /tmp

    pi@raspberrypi / $ cd /tmp/
    pi@raspberrypi /tmp $ ls
    bottle.GFWtyR.lock  omxplayerdbus.pi  omxplayerdbus.pi.pid  screenly_viewer.log  ssh-UOTZig0UrsOY
    pi@raspberrypi /tmp $ cat screenly_viewer.log
    pi@raspberrypi /tmp $

vpetersson, 2015-02-02:
Can you give me the output of the following:

    $ ls -lah /tmp/

    $ ps faux

    $ cat /var/log/supervisor/*.log

craigpelican, 2015-02-02:
pi@raspberrypi ~ $ ls -lah /tmp/
total 32K
drwxrwxrwt  5 root root 4.0K Feb  2 16:26 .
drwxr-xr-x 22 root root 4.0K Jan  1  1970 ..
-rw-------  1 pi   pi      0 Feb  2 16:55 bottle.GFWtyR.lock
drwxrwxrwt  2 root root 4.0K Feb  2 16:12 .ICE-unix
srwxr-xr-x  1 pi   pi      0 Feb  2 16:12 .menu-cached-:0-pi
-rw-r--r--  1 pi   pi     73 Feb  2 16:14 omxplayerdbus.pi
-rw-r--r--  1 pi   pi      5 Feb  2 16:14 omxplayerdbus.pi.pid
srwxr-xr-x  1 pi   pi      0 Feb  2 16:12 .pcmanfm-socket--0-pi
-rw-r--r--  1 pi   pi      0 Feb  2 16:12 screenly_viewer.log
drwx------  2 pi   pi   4.0K Feb  2 16:12 ssh-UOTZig0UrsOY
-r--r--r--  1 root root   11 Feb  2 16:12 .X0-lock
drwxrwxrwt  2 root root 4.0K Feb  2 16:12 .X11-unix
pi@raspberrypi ~ $ ps faux
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         2  0.0  0.0      0     0 ?        S    16:12   0:00 [kthreadd]
root         3  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [ksoftirqd/0]
root         5  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [kworker/0:0H]
root         7  0.0  0.0      0     0 ?        S    16:12   0:01  \_ [rcu_preempt]
root         8  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [rcu_sched]
root         9  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [rcu_bh]
root        10  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [migration/0]
root        11  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [migration/1]
root        12  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [ksoftirqd/1]
root        14  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [kworker/1:0H]
root        15  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [migration/2]
root        16  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [ksoftirqd/2]
root        17  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kworker/2:0]
root        18  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [kworker/2:0H]
root        19  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [migration/3]
root        20  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [ksoftirqd/3]
root        22  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [kworker/3:0H]
root        23  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [khelper]
root        24  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kdevtmpfs]
root        25  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [netns]
root        26  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [perf]
root        27  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [khungtaskd]
root        28  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [writeback]
root        29  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [crypto]
root        30  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [bioset]
root        31  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [kblockd]
root        32  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kworker/0:1]
root        33  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [rpciod]
root        34  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kswapd0]
root        35  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [fsnotify_mark]
root        36  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [nfsiod]
root        42  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [kthrotld]
root        43  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [VCHIQ-0]
root        44  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [VCHIQr-0]
root        45  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [VCHIQs-0]
root        46  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [iscsi_eh]
root        47  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [dwc_otg]
root        48  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [DWC Notificatio]
root        50  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [VCHIQka-0]
root        51  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [SMIO]
root        52  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [deferwq]
root        53  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kworker/u8:2]
root        54  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [mmcqd/0]
root        55  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [jbd2/mmcblk0p6-]
root        56  0.0  0.0      0     0 ?        S<   16:12   0:00  \_ [ext4-rsv-conver]
root        60  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kworker/1:1]
root        73  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kworker/3:1]
root        74  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kworker/2:1]
root      1869  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kworker/3:2]
root      1873  0.0  0.0      0     0 ?        S    16:12   0:00  \_ [kworker/1:2]
root      2394  0.0  0.0      0     0 ?        S    16:13   0:00  \_ [jbd2/mmcblk0p3-]
root      2395  0.0  0.0      0     0 ?        S<   16:13   0:00  \_ [ext4-rsv-conver]
root      2400  0.0  0.0      0     0 ?        S<   16:13   0:00  \_ [kworker/0:1H]
root      2444  0.0  0.0      0     0 ?        S    16:20   0:00  \_ [kworker/0:0]
[... 544 lines cut ...]
2015-02-02 16:04:05,155 CRIT Supervisor running as root (no user in config file)
2015-02-02 16:04:05,157 WARN Included extra file "/etc/supervisor/conf.d/screenly.conf" during parsing
[... cut ...]
2015-02-02 16:12:19,260 WARN cElementTree not installed, using slower XML parser for XML-RPC
2015-02-02 16:12:19,260 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2015-02-02 16:12:19,264 INFO daemonizing the supervisord process
2015-02-02 16:12:19,266 INFO supervisord started with pid 2149
2015-02-02 16:12:20,295 INFO spawned: 'screenly' with pid 2217
2015-02-02 16:12:21,327 INFO success: screenly entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
pi@raspberrypi ~ $

vpetersson, 2015-02-03:
There are a number of errors, but I'm not sure exactly what the root cause is. I hope to have my v2 units in the next few days and will do some further testing then.

craigpelican, 2015-02-03:
Great thanks.
I did, just out of interest, try the current OSE image file. Fails to boot as expected, hangs on the coloured square.
I also tried again with the NOOBS setup as per the raspberrypi.org website as well as the raspbian image file. Same results as above; screenly not loading on boot.
If theres anything else you want me to try, let me know.

jpsingleton, 2015-02-07:
I've just got my v2 so will try this out too. The performance of http://shopify.github.io/dashing/ was just too slow on the v1.

vpetersson, 2015-02-07:
@jpsingleton Yeah that's pretty heavy JS stuff. Will be interesting to see how that plays out on the new version.

ghholm, 2015-02-14:
Just tried to install Screenly the manual way on this NOOBS install of Raspbian on RPI2.

It doesn't seem to start up at all. Had to run xloader.sh from terminal and after some time it showed the splashscreen. Then by some odd miracle is started to show content. But only in that little window and not as fullscreen.

[image]

vpetersson, 2015-02-15:
@ghholm We don't support Noobs. We only support *pure* Raspbian. I don't know the amount of work it would take to make it work on Noobs, but it isn't on our roadmap.

pelicanmedia, 2015-02-16:
@vpetersson I also tried with the pure Raspbian image. But booted straight to the desktop.
I didnt run xloader.sh manually. BUT from seeing the Terminal window in @ghholm's picture...

    pi@raspberrypi ~/screenly/misc $ ./xloader.sh
    uxbl-core: no process found
    omxplayer: no process found
    omxplayer.bin: no process found

Would this indicate that they are not being run on boot ?

vpetersson, 2015-02-16:
@pelicanmedia Those are actually not *errors* per see, as that's just part of the cleanup-process.

I'm currently traveling, so I don't have access to a box to do an installation from scratch. My gut feeling tells me that they've done some changes to the X11 boot process which causes the modifications to not take.

Could you post the output of the installation script?

ghholm, 2015-02-16:
Installed the pure Raspbian instead, only difference I did now vs Noobs was adding sudo at the beginning for installing, but still same result as the screenshot above:

    pi@raspberrypi ~ $ sudo curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash
    Installing Screenly OSE
    Updating system package database...
    Upgrading the system...
    (This might take a while.)
    Installing dependencies...
    grep: /etc/init.d/mathkernel: No such file or directory
    Extracting templates from packages: 100%
    /run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV invocation.
    /run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV invocation.
    Downloading Screenly-OSE...
    remote: Counting objects: 3033, done.
    remote: Total 3033 (delta 0), reused 0 (delta 0)
    Receiving objects: 100% (3033/3033), 1.11 MiB | 751 KiB/s, done.
    Resolving deltas: 100% (1659/1659), done.
    Installing more dependencies...
    Adding Screenly to X auto start...
    Increasing swap space to 500MB...
    Adding Screenly's config-file
    Enabling Watchdog...
    [ ok ] Stopping watchdog keepalive daemon....
    [ ok ] Starting watchdog daemon....
    Adding Screenly to autostart (via Supervisord)
    Making modifications to X...
    Quiet the boot process...
    Assuming no errors were encountered, go ahead and restart your computer.

ghholm, 2015-02-16:
And when I add an video asset this happens:

    pi@raspberrypi ~/screenly $ ./viewer.py&
    [1] 3130
    pi@raspberrypi ~/screenly $ Traceback (most recent call last):
      File "./viewer.py", line 364, in <module>
        main()
      File "./viewer.py", line 360, in main
        asset_loop(scheduler)
      File "./viewer.py", line 302, in asset_loop
        view_video(uri, asset['duration'])
      File "./viewer.py", line 204, in view_video
        if not run.exit_code == 0:
      File "/usr/local/lib/python2.7/dist-packages/sh.py", line 326, in exit_code
        self.wait()
      File "/usr/local/lib/python2.7/dist-packages/sh.py", line 295, in wait
        self._handle_exit_code(self.process.wait())
      File "/usr/local/lib/python2.7/dist-packages/sh.py", line 309, in _handle_exit_code
        self.process.stderr
    sh.ErrorReturnCode_127: 

      RAN: '/usr/bin/timeout 121 mplayer /home/pi/screenly_assets/8e9a3303bbe848f989bdd710833fabcd -nosound'

      STDOUT:

      STDERR:
    /usr/bin/timeout: failed to run command `mplayer': No such file or directory

I guess pure or noobs version of Raspbian is lacking some config and default packages. I though the manual Screenly installer would take care of this ;)

vpetersson, 2015-02-16:
Yeah, looks like something has changed that breaks the logic in `viewer.py`. Hard to troubleshoot remotely. Will have to wait until I get back and I'll take a closer look.

WimRoose, 2015-02-17:
@ghholm: the mplayer error is thrown because new Pi uses armv7 instead of armv6. See code in view_video in viewer.py.

@vpetersson: viewer.py (or actually xloader) fails to start because lxsession is started with LXDE-pi as config option on latest rasbian image. As this folder doesn't exist, X is started with default LXDE settings

vpetersson, 2015-02-17:
@WimRoose Very good insight. I will have to take a stab at fixing this in the next few weeks.

ghholm, 2015-02-19:
@WimRoose Thank you, that seemed to fix it. I replaced all the lxde in the install.sh file with lxde-pi instead. Then fixed viewer.py file with arch=armv71. Now it's starting up like a charm and playing video asset etc in fullscreen as it should ;)

vpetersson, 2015-02-19:
@ghholm Any chance you could post a pull request for that and we can get it merged in?

ghholm, 2015-02-19:
@vpetersson My first time pull request :) Had to double check if I had done it right the first time and noticed one error in the install.sh on 2nd try. But managed to fix that. Some symbolic link that got broken (screenly/misc/lxde-rc.xml). Fix = not adding -pi to that one ;)

vpetersson, 2015-02-20:
Awesome! Thanks.

pelicanmedia, 2015-02-20:
Great work guys. Can't wait to have a play on the super-speedy Pi ;)

anaglyph, 2015-02-26:
I have installed a fresh copy of:
RASPBIAN Debian Wheezy
Version:February 2015 
Release date:2015-02-16 
on a new RPi 2 but appear to be getting a fatal error when trying to install ScreenlyOSE

 ```
pi@raspberrypi ~ $ curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash
Installing Screenly OSE
Updating system package database...
Upgrading the system...
(This might take a while.)
Installing dependencies...
Extracting templates from packages: 100%
/run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV invocation.
/run/udev or .udevdb or .udev presence implies active udev.  Aborting MAKEDEV invocation.
Downloading Screenly-OSE...
fatal: unable to connect to github.com:
github.com[0: 192.30.252.130]: errno=Connection timed out

Installing more dependencies...
Adding Screenly to X auto start...
Increasing swap space to 500MB...
Adding Screenly's config-file
cp: cannot stat '/home/pi/screenly/misc/screenly.conf': No such file or directory
Enabling Watchdog...
[ ok ] Stopping watchdog keepalive daemon....
[ ok ] Starting watchdog daemon....
Adding Screenly to autostart (via Supervisord)
Making modifications to X...
Quiet the boot process...
Assuming no errors were encountered, go ahead and restart your computer.
```
Now I can `ping 192.30.252.129` and `ping github.com` so I am assuming that there is no firewall issue.
Would anyone have an idea what the problem might be or able to point me at some things to try?

vpetersson, 2015-02-26:
@anaglyph That's odd. It could perhaps have been a temporary network issue. I was able to run the installer on `2015-02-16-raspbian-wheezy.zip` just yesterday when I committed the code.

I would suggest that you wipe the device and try again.

anaglyph, 2015-02-26:
I think I found the solution:
I created a temp directory, and used `git clone https://github.com/wireload/screenly-ose.git` to download the repo. I then changed `git:\\` to `https:\\` the line
 `git clone -q git://github.com/wireload/screenly-ose.git "$HOME/screenly" > /dev/null` 
of install.sh so it read 
`git clone -q https://github.com/wireload/screenly-ose.git "$HOME/screenly" > /dev/null` 
and ran the bash script locally.

vpetersson, 2015-02-26:
@anaglyph Interesting. `https://` might be more reliable than `git://` in this situation. I've pushed a change to master that does just that.

anaglyph, 2015-02-26:
Very kind @vpetersson . Many thanks. I think I will do a fresh install of `2015-02-16-raspbian-wheezy.zip` and start again!

vpetersson, 2015-02-26:
@anaglyph Cool. Let me know how it goes.

closed 2015-02-26 by vpetersson

pelicanmedia, 2015-02-27:
Works perfectly. Well done guys !

jpsingleton, 2015-02-28:
Works great on my pi 2 as well. Installed with NOOBS lite from a desktop shell too. Good job!

SSL still works when enabled and http://dashingdemo.herokuapp.com/sampletv performs quite well, although it could be smoother. This is probably because the browser is still single threaded.

vpetersson, 2015-02-28:
@jpsingleton Cool! Regarding the browser, it's because the browser isn't hardware accelerated. Take a look at #257 for more details.

Cyberprog, 2015-05-15:
Am I right in thinking that the images on the website now support the Raspberry Pi 2 Model B ok? I need to order another couple of Pi's, so figure it would be best to get the newer model as the cost difference is negligible...

vpetersson, 2015-05-15:
@Cyberprog Yes, the current build works on the RasPi 2.

Cyberprog, 2015-05-15:
Excellent, thanks.