Screenly/Anthias #257: Improved web loading - Replace UZBL with Epiphany/Web

vpetersson, 2015-02-04:
In theory, it would be possible to replace UZBL with [Epiphany/Web](http://www.raspberrypi.org/web-browser-beta/). There are big benefits with doing so, as we'll get support for hardware acceleration and better and hence better experience.

Swapping out the browser itself is fairly trivial. The problem is remote controlling it. 

In theory, Epiphany/Web supports DBUS for these actions, but the support seems broken. I've opened a [ticket](https://bugzilla.gnome.org/show_bug.cgi?id=743982) for that. 

Spinning up the browser itself works great. All we need to do is to run:

    #!/bin/bash
    export DISPLAY=:0
    mkdir -p /home/pi/tmp
    epiphany-browser --display=:0  --profile=/home/pi/tmp -a http://wireload.net

That should in theory then register with DBUS, which should allow us to remote control it. There's even built-in support for doing this using [Python](http://dbus.freedesktop.org/doc/dbus-python/doc/tutorial.html).

Documentation on the actual DBUS commands are scarce, but there are some initial pointers [here](https://mail.gnome.org/archives/epiphany-list/2007-November/msg00009.html).

peelman, 2015-02-13:
well this is exciting to contemplate...doesn't look like your bug is gaining a lot of traction with them, but hopefully something will shake out of it...

kaihendry, 2015-04-22:
Sadly the accelerated work commissioned by the PI foundation is only for a non-upstreamed & the old libwebkitgtk-3.0. Some more details here: http://archlinuxarm.org/forum/viewtopic.php?f=64&t=8673

vpetersson, 2015-04-22:
Good link, @kaihendry.

tiptronic, 2015-05-24:
@vpetersson I just played with Screenly and - compared to some other solutions - UZBL-performance is better than Firefox, Chrome and some others (Midori comes close).
How would I define Epiphany as default browser for Screenly? (I don't care too much about remote control, as I just show one webpage, which loads it's screens by itself (for now)).

vpetersson, 2015-05-25:
@tiptronic It's unfortunately not that easy. The remote control feature is critical to Screenly as it is how Screenly interacts with the browser.

That said, you could in theory replace the browser [here](https://github.com/wireload/screenly-ose/blob/master/viewer.py#L140), but it would only make Screenly load the browser. You could perhaps hardcode a URL there but it won't be able to do anything else from that point on.

tiptronic, 2015-05-25:
@vpetersson Thx for the hint.  Do you mean the latest from git or just updating Screenly through your update script (does it pull from Git?)

Over the weekend I did a simple test:
 - installed a fresh Raspbian, some tools and created a kiosk-stlye system, which uses Epiphany to load my pages. 
- I connected a HTML5/CSS3 app, which works as configurable signage backend.
- I ran it against the current Screenly OSE distro (from your website). And you know what? Even though I overclock the RPI 2 (running Screenly) to 1MHz, the bare-bone Epiphany ran *much* better (and I mean really much). It even runs complex CSS3 animations and transforms (somewhat stutterfree ;)).

The only thing I couldn't get to work smoothly (yet) is video (I tried the same bigbuckbunny.mp4 as Screenly, although I streamed it over my LAN). (I have yet to find out how to play video locally while my app sits on another server)

So that being said, if you try this out for yourself, I'm pretty sure you see what I mean.

vpetersson, 2015-05-25:
> Do you mean the latest from git or just updating Screenly through your update script (does it pull from Git?)

Yeah, the update script pulls from Git, so should be the same.

> The bare-bone Epiphany ran much better (and I mean really much).
> It even runs complex CSS3 animations and transforms (somewhat stutterfree ;)).

Yeah, that's unfortunately not all that surprising. UZBL is far behind, but since the other browsers are lacking the remote control ability, we're stuck with it for now. The only way around that would be to use the DBUS feature, but it doesn't seem to work very well.

tiptronic, 2015-05-25:
Did you file a FR with the Epiphany devs for Remote Control?

vpetersson, 2015-05-25:
Yes, it's in the ticket above (https://bugzilla.gnome.org/show_bug.cgi?id=743982).

tiptronic, 2015-05-25:
Hm... it sounds they won't fix this. Did you try to remove g_application_set_flags()?

vpetersson, 2015-05-25:
No, didn't have a chance to play with that yet.

radnotiz, 2015-06-03:
Hi there, you mentioned problems with remote controlling the browser. Would it be possible to drive it through the WebDriver API? http://www.w3.org/TR/2013/WD-webdriver-20130117/ (Unfortunately I need one of the mainstream browsers to present my content on a screen and found UZBL's performance and feature set being not sufficient.)

vpetersson, 2015-06-03:
@radnotiz That's an interesting idea. The W3C thing appears to be just a document draft without any implementation (but I could be wrong). Selenium's [Web Driver](http://docs.seleniumhq.org/docs/03_webdriver.jsp) on the other hand is fully implemented and could work. It would still require a bit of testing to determine if it is a viable route and what the support on the Pi is, but it is an interesting idea.

tiptronic, 2015-06-03:
@vpetersson Here's an example implementation on nodejs:
https://github.com/groupon/webdriver-http-sync

vpetersson, 2015-06-03:
Interesting, but we'd need a Python version.

vpetersson, 2015-10-05:
Unfortunately this browser doesn't seem to be very actively developed:
http://archive.raspberrypi.org/debian/pool/main/e/epiphany-browser/

There was however *some* activity on the issue I created upstream (tl;dr will accept a fix):
https://bugzilla.gnome.org/show_bug.cgi?id=743982

tiptronic, 2015-10-05:
From what I read they currently concentrate on bug-fixes and performance improvement:
<pre>
There are updates to several of the applications that used to come with Raspbian. There are new versions of Scratch, Sonic Pi, and the Epiphany web browser; none of these have changed fundamentally in operation, but they all include bug fixes and performance improvements.
</pre>

It's out of the scope of my experience, what's needed to make that work, but I'd still vote for a test ;)

vpetersson, 2015-10-05:
Yeah, I want to take a closer look at this in Sprint 3.

vpetersson, 2016-12-13:
Closing out in favor of #527. The Epiphany/Web appears to be dead.

closed 2016-12-13 by vpetersson