Screenly/Anthias #258: Stop running at random asset

willem2668, 2015-02-04:
2-3 months without any issue, since last update Screenly is always halting just before "Received on 200-status" (see log).
The display keeps showing the last asset.

Things I tried:
* reboot
* cold reboot
* sudo shutdown -rF now

Below the log (/tmp/screenly_viewer.log)

```
Wed, 04 Feb 2015 12:41:12 Loading browser...
Wed, 04 Feb 2015 12:41:12 Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 2534.
Wed, 04 Feb 2015 12:41:41 Scheduler init
Wed, 04 Feb 2015 12:41:41 update_playlist
Wed, 04 Feb 2015 12:41:41 Generating asset-list...
Wed, 04 Feb 2015 12:41:41 generate_asset_list deadline: 2015-02-08 13:00:00
Wed, 04 Feb 2015 12:41:42 update_playlist done, count 14, counter 0, index 0, deadline 2015-02-08 13:00:00
Wed, 04 Feb 2015 12:41:42 Entering infinite loop.
Wed, 04 Feb 2015 12:41:42 Last update: 2015-01-29 23:23:37.425956
Wed, 04 Feb 2015 12:41:45 Received on 200-status
Wed, 04 Feb 2015 12:41:45 get_next_asset
Wed, 04 Feb 2015 12:41:45 refresh_playlist
Wed, 04 Feb 2015 12:41:45 refresh: counter: (0) deadline (2015-02-08 13:00:00) timecur (2015-02-04 11:41:45.079167)
Wed, 04 Feb 2015 12:41:45 get_next_asset after refresh
Wed, 04 Feb 2015 12:41:45 get_next_asset counter 0 returning asset 1 of 14
Wed, 04 Feb 2015 12:41:45 Showing asset WIK logo (image)
Wed, 04 Feb 2015 12:41:45 Asset URI /home/pi/screenly_assets/2031b7082e774f87a4c686c3d72602d2
Wed, 04 Feb 2015 12:41:45 Already showing file:///tmp/screenly_html/black_page.html, keeping it.
Wed, 04 Feb 2015 12:41:45 Sleeping for 25
Wed, 04 Feb 2015 12:42:10 Last update: 2015-01-29 23:23:37.425956
Wed, 04 Feb 2015 12:42:11 Received on 200-status
Wed, 04 Feb 2015 12:42:11 get_next_asset
Wed, 04 Feb 2015 12:42:11 refresh_playlist
Wed, 04 Feb 2015 12:42:11 refresh: counter: (0) deadline (2015-02-08 13:00:00) timecur (2015-02-04 11:42:11.103476)
Wed, 04 Feb 2015 12:42:11 get_next_asset after refresh
Wed, 04 Feb 2015 12:42:11 get_next_asset counter 0 returning asset 2 of 14
...

...
Wed, 04 Feb 2015 12:53:06 get_next_asset counter 0 returning asset 12 of 14
Wed, 04 Feb 2015 12:53:06 Showing asset Chiro Mosselen (image)
Wed, 04 Feb 2015 12:53:06 Asset URI /home/pi/screenly_assets/cb64cb41452446e18bea72047301d5fc
Wed, 04 Feb 2015 12:53:06 Already showing file:///tmp/screenly_html/black_page.html, keeping it.
Wed, 04 Feb 2015 12:53:06 Sleeping for 25
Wed, 04 Feb 2015 12:53:31 Last update: 2015-01-29 23:23:37.425956
Wed, 04 Feb 2015 12:53:32 Received on 200-status
Wed, 04 Feb 2015 12:53:32 get_next_asset
Wed, 04 Feb 2015 12:53:32 refresh_playlist
Wed, 04 Feb 2015 12:53:32 refresh: counter: (0) deadline (2015-02-08 13:00:00) timecur (2015-02-04 11:53:32.384506)
Wed, 04 Feb 2015 12:53:32 get_next_asset after refresh
Wed, 04 Feb 2015 12:53:32 get_next_asset counter 0 returning asset 13 of 14
Wed, 04 Feb 2015 12:53:32 Showing asset Eetdagen KlimOp 2015 (image)
Wed, 04 Feb 2015 12:53:32 Asset URI /home/pi/screenly_assets/480dae8ead65482a98c19d3dbe4b3109
Wed, 04 Feb 2015 12:53:32 Already showing file:///tmp/screenly_html/black_page.html, keeping it.
Wed, 04 Feb 2015 12:53:32 Sleeping for 25
Wed, 04 Feb 2015 12:53:57 Last update: 2015-01-29 23:23:37.425956
```

vpetersson, 2015-02-04:
That's a typo. I've pushed a fix. Should be 'Received *non* 200-status', meaning that the webpage it's trying to display is having issues. Hence, it's not an error in Screenly, but rather with the web-page.

closed 2015-02-04 by vpetersson

CrownWine, 2016-04-20:
Hi I am having that same issue in the last version 2016-04-18, the black page came and keep in that forever

willem2668, 2016-04-20:
> Hi I am having that same issue in the last version 2016-04-18, the black page came and keep in that forever

Above issue is dated February 2015.
Didn't find the solution, but since I installed Screenly on a new SD card, until now the system is running (with regular updates) like a charm :-)

vpetersson, 2016-04-21:
@CrownWine Please paste the logs. It's most likely a website that doesn't respond with a proper HTTP status code.

CrownWine, 2016-04-21:
This is the log:

[image]

vpetersson, 2016-04-21:
Looks like it is working according to the logs. Try enabling debug logging.