Screenly/Anthias #259: Screenly adding 20s to set duration.

dtdickman, 2015-02-05:
Hi there,

I've just implemented Screenly OSE at work and we are having some issues with asset durations. It seems that no matter what time I set an asset to, an additional 20s is always added. 

The image below shows 2 active assets. For the purpose of illustrating my issue, one asset is set with a duration of 10s and the other to 2s. Without fail, the slide never changes for additional 20s in addition to whatever the slides duration is.

Is anyone else having this issue? 

[image]

vpetersson, 2015-02-05:
Hmm that's odd. Some initial questions:

 * Are you running anything else on the device beyond Screenly? 
 * What kind of SD card are you using (class, brand, model)?

dtdickman, 2015-02-05:
Actually I just resolved the issue by running a software update. Thanks for the quick reply!

vpetersson, 2015-02-06:
Great!

closed 2015-02-06 by vpetersson