Screenly/Anthias #260: Password protect the Screenly admin interface

jpsingleton, 2015-02-07:
Password protection for the Screenly admin interface seems to be a common request (https://github.com/wireload/screenly-ose/issues/121 https://github.com/wireload/screenly-ose/issues/133) but I couldn't find an issue just dealing with it on it's own. There is this PR https://github.com/wireload/screenly-ose/pull/224 but that seems to have stalled.

We currently have pi screens running on our guest WiFi network (so credentials being recovered from the SD card isn't a big deal). However, this means that if we used screenly then anyone could change the content using the open admin interface. Currently we use a home brew version that reads a URL from the FAT partition to show a graph (https://shutdownscanner.com/home/about#digital-sign) and has no web interface (we use SSH and VNC). There is a simple setup guide (https://shutdownscanner.com/blog/posts/raspberry-pi-digital-sign) but a lot of this work has already been done by screenly.

I see a couple of options:

- **HTTP basic authentication** (really this needs a certificate for TLS/SSL too but on an internal network this isn't really required as long as the credentials are unique)
- **HTTP digest authentication** (which doesn't send the password as it uses a hash)
https://en.wikipedia.org/wiki/Digest_access_authentication

vpetersson, 2015-02-07:
@jpsingleton Yeah, this stalled per the feedback. Hence the PR requires a bit more work. (ping @axel-b). Would love to see that one merged tho.

jpsingleton, 2015-02-07:
That diff looks pretty big for what should be fairly simple (likely due to the LDAP support which probably isn't really needed). http://cork.firelet.net/ and https://pypi.python.org/pypi/bottle-auth/ both look promising. Or this could be done at the web server level ignoring the framework. I'll see what I can do.

Ideally the password should be securely hashed on disk but then you'd need an admin interface (or at least a tool) to set it up. If the web interface allows setting up a password then users could easily get locked out. Maybe there should be a default (randomly generated) password?

vpetersson, 2015-02-07:
> Ideally the password should be securely hashed on disk but then you'd need an admin interface

I'm honestly not too concerned about the storage on disk. Sure, it should be hashed etc of course, but if someone gains that level of control of the box, it's pretty much compromised. 

I'm far more concerned about the credentials in flight. Hence the credential feature should only work combined with the https support.

>  If the web interface allows setting up a password then users could easily get locked out.

I think the best way to mitigate this is to have a cli tool for changing the password.

jpsingleton, 2015-02-07:
OK, sounds like the simplest thing that can possibly work is a password configured in plain text on disk. That makes it easy to set up without a web interface or tool and makes it hard to get locked out.

Setting up HTTPS without just using a simple self-signed certificate is fairly involved. However, by using digest (instead of basic) authentication it can work over plain HTTP without revealing any credentials.

vpetersson, 2015-02-07:
I don't think they should be stored in plain text per se, but something simple like SHA etc should do. In fact, we could even use the format used by Apache's basic auth, which I know other tools use too. That way we can even re-use other tools for password update/recovery (and would allow multiple users).

In terms generating self-signed certs, there's a script for that available [here](https://github.com/wireload/screenly-ose/blob/master/misc/enable_ssl.sh).

axel-b, 2015-02-08:
LDAP support is the one and only reason why I did the work -- but if that
can be achieved in an easier way, great!
Anyway, my personal involvement with running screenly-ose  stopped a while
ago.

Axel.

axel-b, 2015-02-08:
> @jpsingleton <https://github.com/jpsingleton> Yeah, this stalled per the
> feedback. Hence the PR requires a bit more work. (ping @axel-b
> <https://github.com/axel-b>). Would love to see that one merged tho.
Sorry, you will have to do the work  if you care enough -- I'm no longer
involved in running screenly-ose.
Axel.

vpetersson, 2015-02-08:
Hi @axel-b. Understood. Thanks for the work you have done though. Hopefully we can re-use some of the code you've written.

jpsingleton, 2015-02-08:
If it's worth doing then it's worth doing properly. Securely hash the passwords with a salt and only use HTTPS (a self-signed certificate is better than nothing).

It looks like you're still using the built in development web server. Is that right? If upgraded to a real server then it makes a lot of this easier.

I'll open a PR and have a go when I find some time.

vpetersson, 2015-02-08:
@jpsingleton That's correct. Switching to Gunicorn or something makes sense. I don't want to introduce a full-blown web-server (like Apache or Nginx), as it would carry a fair amount of overhead.

Mugiwara75, 2015-02-12:
Hi all,

thanks for "screenly ose" I've just tested it and it's work perfectly. Very good work !

I just want to follow this thread because I'm very interested to have a login/password for the webadmin. Do you have an approximativly date of implementation ?

Thanks

vpetersson, 2015-02-12:
Hi @Mugiwara75. Nobody is actively working on this to my knowledge.

Mugiwara75, 2015-02-12:
Hi thanks for your answer, do you know if screenly company project to add this functionnality ?

vpetersson, 2015-02-12:
@Mugiwara75 This already supported in Screenly Pro, but we have no plans right now to allocate resources for porting this to Screenly OSE.

Mugiwara75, 2015-02-12:
@vpetersson oh too bad for me :-(. I want to protect the webadmin just by a login and password :-(

jpsingleton, 2015-02-12:
I have a version of this working with the admin pages (all pages except the splash screen) and API calls protected by password authentication over SSL. It works pretty well but I'm not sure how to make it easily configurable.

vpetersson, 2015-02-12:
@jpsingleton Could you do a pull request for that? Perhaps we can get that sorted together.

jpsingleton, 2015-02-12:
Will do. I've just used the built in auth_basic in bottle. This would work if you always want to enforce passwords by default but I can't find an easy way to not prompt the user for credentials even when none are set.

vpetersson, 2015-02-12:
Oh ok. So when we started working on Screenly, there was no such built-in support. If there is now, we should definitely use it instead of re-inventing the wheel.

axel-b, 2015-02-12:
In my code the prompting was conditional, don't know of it might give
inspiration.

richard-to, 2015-02-13:
@jpsingleton, Looks like you're using the auth_basic decorator. I wonder if you can create your own decorator and then add a conditional statement like @axel-b  suggested. You'd then just need to manually call the auth_basic decorator function, I think.

jpsingleton, 2015-02-13:
Yes, I am using the auth_basic decorator and it works great for my use case. If Screenly OSE should always be authenticated (and only over HTTPS) then let me know and I'll make a PR. Otherwise I don't think it's the correct solution without more work. 

Most routers come with default credentials to the web admin interface which can be changed so there is precedent for this pattern. The pi user password needs changing anyway. It should be fairly easy to authenticate against the pi (or other) user in the `/etc/shadow` file using https://pythonhosted.org/passlib/.

vpetersson, 2015-02-13:
Hmm, would you really authenticate the web interface using a system user? That doesn't seem ideal.

jpsingleton, 2015-02-13:
No, it's not a good idea but it is simple (although it would probably have to run as root which is another issue). This isn't how I've set mine up. PassLib can be used to read Apache files just as easily (https://pythonhosted.org/passlib/lib/passlib.apache.html).

vpetersson, 2015-02-13:
That seems like a better approach.

ocarino, 2015-05-08:
Well, I am not sure why we should solve all problems at once (this only makes sure that it will never happen). So, I did not find the code of @axel-b so if it is here somewhere it should be pretty easy to adapt it.

If not: It took me about 1.5 hours (and I am very new to python) to implement the following:

- use basic auth for every view in server.py
- set two variables in setup.py, called "login_user" and "login_pw".

It is really easy because bottle already has @auth_basic...

So, yes, I have the password in cleartext in setup.py and I do not have SSL encryption. This is not good, but it is already way better than having nothing. My screenly raspberries run in a closed network environment, so sniffing is probably no problem and if someone can read the source code on the raspberry already I do not have anything more to loose...

If someone cares I could provide my changes. Right now the server always asks for basic authentication (even if the variables login_user and login_pw are not set) so there is much room for improvement. Any takers?

Cheers, Olaf

jpsingleton, 2015-05-08:
Sounds like you have a very similar code base to what I've hacked together. If I can find some time to polish it up then I'll open a PR but right now it's not of high enough quality to merge.

vpetersson, 2015-05-08:
@ocarino A lot of things have been added to Bottle since we started working on Screenly OSE. Built-in authentication is one of them. 

Also, the reason for not adding a hacking solution before adding SSL (which we now support) was that the a false sense of security is worse than no security.

ocarino, 2015-05-08:
@vpetersson  :  Did not notice that SSL is already there (but maybe I just searched on web pages and not in the source, I am running OSE). So, I do have a lousy solution that works for me and I will be happy to update if you have a good solution. :-)

> Also, the reason for not adding a hacking solution before adding SSL (which we now support)
> was that a false sense of security is worse than no security.

What would be more insecure than having no authentication at all? In the actual situation nobody has security (independent of caring about that or not). Anyway, I do understand that you (as a distributor/developer/vendor) want a good solution in your product and not a crude hack.

Cheers, Olaf

vpetersson, 2015-05-08:
@ocarino I'd say sending credentials in plain text is a false sense of security :)

In any case, here's the script that enables SSL:
https://github.com/wireload/screenly-ose/blob/master/misc/enable_ssl.sh

CrossCutter, 2015-05-16:
@ocarino I would appreciate a copy of your changes - I am of the same mind as you that basic auth is much much better than the zero we have right now :-)

jpsingleton, 2015-05-16:
If there is demand then I'll push what I've got so far. Not ready for a
pull request yet but fine if you want to manually hack credentials directly
in the python. Works for me :).

kpdillon, 2015-05-16:
I am interested too!

jpsingleton, 2015-05-18:
OK @CrossCutter & @kpdillon, here you go!

The updated code with basic auth is [here](https://github.com/jpsingleton/screenly-ose) (instructions at the top of the README).
You **really** should change the credentials from the default [here](https://github.com/jpsingleton/screenly-ose/blob/master/server.py#L45) (and enable SSL).

Enjoy!
:shipit:

closed 2015-05-18 by jpsingleton

kpdillon, 2015-05-18:
Thank you so much!  Just to be clear - does SSL have to be enabled for this
to work?  I know it's recommended but wondering if it will work with it?

jpsingleton, 2015-05-18:
No, it's not required. However, the password will be sent over the network in plain text if not enabled.

kpdillon, 2015-05-18:
Yep, understood - thank you.  I am only accessing it over an SSH tunnel so
it will be encrypted that way.  This will help in stopping anyone from
accessing if they are on the same VLAN.

CrossCutter, 2015-05-18:
@jpsingleton Fantastic thanks

Frankstar, 2015-12-21:
any plan to add @jpsingleton work into the master ? Or mayb we should create a script which adds hes work ? (as a workaround)

vpetersson, 2015-12-21:
It requires a fair amount of work to integrate into the current code base (and add tests). It is however on the roadmap.

jpsingleton, 2015-12-21:
I wouldn't recommend merging in my workaround. It really is a hack which is why I haven't raised a PR. I would build a more robust implementation if the idea was to merge upstream.

I've pulled in the latest changes from the upstream master so you can [install](https://www.screenlyapp.com/ose/) on a pi by running:

```
$ curl -sL https://raw.github.com/jpsingleton/screenly-ose/master/misc/install.sh | bash
```

Or you can just overwrite `server.py` with the file from my repo as there weren't any conflicts.

vpetersson, 2015-12-21:
@jpsingleton Sorry, I mixed this up with @axel-b's PR (#224).

muwml, 2016-01-06:
Disappointing @vpetersson that a feature like this does not make it to OSE.  Thank you @jpsingleton for your work.

vpetersson, 2016-01-06:
@muwml We're not rejecting this feature. There is a pull request as mentioned above that adds this. However, it would require additional work to be integrated into our system (which is tracked in #306).