Screenly/Anthias #261: JPEG Display

fenix687, 2015-02-09:
I have uploaded several JPEGs 1920x1080px.  They are not outputting at that resolution, they appear "under-scanned".  I have tried to use the HDMI out and the Compostite out with the same issue.  

any insight would be greatly appreciated.  

thank you

jpsingleton, 2015-02-13:
There are a few reasons this could occur. Are you seeing a black border around the image or is it cropped and missing its edges?

Things to try:
- enable/disable overscan on the pi (in config.txt or using raspi-config)
- enable/disable overscan in the TV settings (if available) 
 - this can be called many things depending on the make e.g. `Just Scan`
 - it may be set as part of an aspect ratio e.g. `16:9` or `1:1`

fenix687, 2015-02-18:
@jpsingleton There is no over or under scanning on the TV or raspberry pi.  the boot and splash are correct proportion and fill the screen.  nothing is getting chopped off its just getting squeezed down to an unusual resolution. 

thanks

vpetersson, 2015-02-18:
Very odd. What happens if you try a web page or non-JPEG image?

fenix687, 2015-02-21:
Web pages are the same story, looks under scanned.  videos are full screen and look correct.

vpetersson, 2015-02-21:
If videos look fine, and web pages and images are not, then it's definitely an overscan issue. Try modifying `config.txt` and tweak the overscan settings.

fenix687, 2015-02-21:
all of that is commented out on the config.txt .  if i adjust it, it messes up the video assets.

vpetersson, 2015-02-21:
>  all of that is commented out on the config.txt . if i adjust it, it messes up the video assets

No, video playback is using framebuffer output, which doesn't use the overscan settings in `config.txt`.

fenix687, 2015-02-21:
sorry, I was looking at a different Pi.  I have like 20 running different things.  

Yes, adjusting overscan seems to have fixed it.  So sorry for the confusion. @vpetersson  for the win. 

thank you

vpetersson, 2015-02-21:
Great. Glad you got it working in the end.

closed 2015-02-21 by vpetersson