Screenly/Anthias #263: Server Error: Could not retrieve file. Check the asset URL.

PP2000, 2015-02-16:
I'm trying to display my site but I'm getting an error message: Server Error: Could not retrieve file. Check the asset URL.

curl -I "http://www.dachdeckerbetrieb-waechter.de/" seems ok
Here's the result:
--------------------------
pi@screenly ~ $ curl -I "http://www.dachdeckerbetrieb-waechter.de/"
HTTP/1.1 200 OK
Date: Mon, 16 Feb 2015 14:09:09 GMT
Server: Apache
P3P: CP="NOI ADM DEV PSAi COM NAV OUR OTRo STP IND DEM"
Cache-Control: no-cache
Pragma: no-cache
Set-Cookie: 9aac6915d20f289e0835ea1b1ce0af10=ofmvvhrgkmjssgjgkjbe2i7nc5; path=/
Vary: Accept-Encoding
Content-Type: text/html; charset=utf-8
-----------------------

Any ideas?

Thanks.

vpetersson, 2015-02-16:
Hmm. I was able to reproduce this issue on the [demo site](http://ose.demo.screenlyapp.com/). 

Not sure exactly what's going on, but could you check the logs?

PP2000, 2015-02-17:
Logs after trying: 

/tmp/screenly_viewer.log
Tue, 17 Feb 2015 06:03:21 Loading browser...
Tue, 17 Feb 2015 06:03:21 Browser loading http://0.0.0.0:8080/splash_page. Running as PID 2628.
Tue, 17 Feb 2015 06:04:25 Generating asset-list...
Tue, 17 Feb 2015 06:04:26 Showing asset Screenly Tour (webpage)
Tue, 17 Feb 2015 06:04:27 current url is http://www.screenlyapp.com/tour.html
Tue, 17 Feb 2015 06:04:27 Sleeping for 60
Tue, 17 Feb 2015 06:05:28 Generating asset-list...
Tue, 17 Feb 2015 06:05:28 Showing asset Screenly Tour (webpage)
Tue, 17 Feb 2015 06:05:28 Sleeping for 60
Tue, 17 Feb 2015 06:06:30 Showing asset Screenly Comparison (webpage)
Tue, 17 Feb 2015 06:06:30 current url is http://www.screenlyapp.com/pricing.html
Tue, 17 Feb 2015 06:06:30 Sleeping for 60
Tue, 17 Feb 2015 06:07:31 Generating asset-list...
Tue, 17 Feb 2015 06:07:32 Showing asset Screenly Tour (webpage)
Tue, 17 Feb 2015 06:07:32 current url is http://www.screenlyapp.com/tour.html
Tue, 17 Feb 2015 06:07:33 Sleeping for 60
Tue, 17 Feb 2015 06:08:34 Showing asset Screenly Comparison (webpage)
Tue, 17 Feb 2015 06:08:35 current url is http://www.screenlyapp.com/pricing.html
Tue, 17 Feb 2015 06:08:35 Sleeping for 60

/tmp/screenly_xloader.log
Disabling screen power savings...
Launching infinite loop...
Cleaning up...

I think that's not the correct logfile?

vpetersson, 2015-02-17:
Ok, here's the traceback:

    127.0.0.1 - - [17/Feb/2015 19:57:36] "GET /api/assets HTTP/1.0" 200 2
    Traceback (most recent call last):
      File "/home/screenly-ose/screenly/server.py", line 209, in api_view
        return make_json_response(view(*args, **kwargs))
      File "/home/screenly-ose/screenly/server.py", line 223, in add_asset
        raise Exception("Could not retrieve file. Check the asset URL.")
    Exception: Could not retrieve file. Check the asset URL.
    127.0.0.1 - - [17/Feb/2015 19:57:46] "POST /api/assets HTTP/1.0" 500 58

I poked around some more, and it appears as it has to do with the website. For some reason, it returns 403 when trying to get the head using `request`:

    >>> import requests
    >>> url = 'http://www.dachdeckerbetrieb-waechter.de/'
    >>> head = requests.head(url, allow_redirects=True, timeout=10)
    >>> head.status_code
    403
    >>> head.content
    ''
Not much we can do unfortunately.

closed 2015-02-17 by vpetersson