Screenly/Anthias #264: Ordering via POST in API doens't work

aqt01, 2015-02-18:
Hi Blessings! 
I have 2 assets and i'm trying to order it by posting their ID and play_order like shown below
http://pastebin.com/u5BrZCR2
Also, i tried with one asset like: 
http://pastebin.com/WUdk9dNZ

I'm getting <200 response> with both examples, but there's no change on the playlist in screenly ose admin interface and neither in /api/assets/

richard-to, 2015-02-18:
I haven't tested this out. I only took a quick look at the code, but I think `play_order` is ignored and not saved to the database when creating or updating an asset. Usually on update, another api request is sent to update the order.

Instead you have to use: `/api/assets/order`

This takes a POST request with the parameter `ids`, which is a comma separated list of asset ids.

You can take a look at the network requests on the Screenly OSE admin to see what it does.

There's probably a reason why play order is set this way. But I don't know the reason.

aqt01, 2015-02-19:
I used firebug in forefox to check,  and is something like

Parametersapplication/x-www-form-urlencoded
ids	e97b1dd4009644df82bdd0b19955f65f,1e0c2dac888c4cfc837f09a98577961a
Source
ids=e97b1dd4009644df82bdd0b19955f65f%2C1e0c2dac888c4cfc837f09a98577961a

I've modified and tested as:
A.- payload = {'ids[]':['1e0c2dac888c4cfc837f09a98577961a','e97b1dd4009644df82bdd0b19955f65f']}
B.- payload = {'ids':['1e0c2dac888c4cfc837f09a98577961a','e97b1dd4009644df82bdd0b19955f65f']}

and sended using: 
r = requests.post('http://192.168.1.21:8080/api/assets/order', data=payload, timeout=4)

But still the same

richard-to, 2015-02-20:
@aqt01 You have the format slightly off. The post param is not in JSON format for this api endpoint. The ids parameter expects a string. The string is then split by comma in the backend.

It should look like this:

    ids=1e0c2dac888c4cfc837f09a98577961a,e97b1dd4009644df82bdd0b19955f65f

closed 2015-09-23 by vpetersson

xstable, 2018-05-01:
Please reopen this Ticket...

I try to reorder the List via POST-ORDER, but in every play_order-field of each asset the value "1" is set:

[image]

And here the result after usin ORDER:
```
[
	{
		"asset_id": "6d6fb0de28da436d99686fa0f312a34b",
		"mimetype": "webpage",
		"name": "http://www.webpage.com/facebook-news/facebook-news-01.htm",
		"end_date": "2019-11-15T20:39:00+00:00",
		"is_enabled": 1,
		"nocache": 0,
		"is_active": true,
		"uri": "http://www.webpage.com/facebook-news/facebook-news-01.htm",
		"duration": "150",
		"play_order": 1,
		"start_date": "2017-11-08T20:39:00+00:00",
		"is_processing": 0
	},
	{
		"asset_id": "cd995762c38649dfadf02f5827a7d05f",
		"mimetype": "webpage",
		"name": "http://www.webpage.com/facebook-news/facebook-news-02.htm",
		"end_date": "2019-11-15T20:40:00+00:00",
		"is_enabled": 1,
		"nocache": 0,
		"is_active": true,
		"uri": "http://www.webpage.com/facebook-news/facebook-news-02.htm",
		"duration": "15",
		"play_order": 1,
		"start_date": "2017-11-08T20:40:00+00:00",
		"is_processing": 0
	},
	{
		"asset_id": "d5b1b6794372412fa81466ecc2c83efe",
		"mimetype": "webpage",
		"name": "http://www.webpage.com/facebook-news/facebook-news-03.htm",
		"end_date": "2019-11-15T20:40:00+00:00",
		"is_enabled": 1,
		"nocache": 0,
		"is_active": true,
		"uri": "http://www.webpage.com/facebook-news/facebook-news-03.htm",
		"duration": "15",
		"play_order": 1,
		"start_date": "2017-11-08T20:40:00+00:00",
		"is_processing": 0
	},
	{
		"asset_id": "1cd74c92c53f4eb7873dcaf4c0c221f5",
		"mimetype": "webpage",
		"name": "http://www.webpage.com/facebook-news/facebook-news-04.htm",
		"end_date": "2019-11-15T20:41:00+00:00",
		"is_enabled": 1,
		"nocache": 0,
		"is_active": true,
		"uri": "http://www.webpage.com/facebook-news/facebook-news-04.htm",
		"duration": "15",
		"play_order": 1,
		"start_date": "2017-11-08T20:41:00+00:00",
		"is_processing": 0
	},
	{
		"asset_id": "dc8eda29281b43e3b52c23bedf740c22",
		"mimetype": "webpage",
		"name": "http://www.webpage.com/facebook-news/facebook-news-05.htm",
		"end_date": "2019-11-15T20:41:00+00:00",
		"is_enabled": 1,
		"nocache": 0,
		"is_active": true,
		"uri": "http://www.webpage.com/facebook-news/facebook-news-05.htm",
		"duration": "15",
		"play_order": 1,
		"start_date": "2017-11-08T20:41:00+00:00",
		"is_processing": 0
	},
	{
		"asset_id": "17ff17c1305847dca44d996a7eff3d14",
		"mimetype": "webpage",
		"name": "http://www.webpage.com/facebook-news/facebook-news-06.htm",
		"end_date": "2019-11-15T20:41:00+00:00",
		"is_enabled": 1,
		"nocache": 0,
		"is_active": true,
		"uri": "http://www.webpage.com/facebook-news/facebook-news-06.htm",
		"duration": "10",
		"play_order": 1,
		"start_date": "2017-11-08T20:41:00+00:00",
		"is_processing": 0
	},
	{
		"asset_id": "youtube_UExUNFNHYmxjRGUtcV9HMndXaWljWTBwazdxZFl3ajFQci41NkI0NEY2RDEwNTU3Q0M2",
		"mimetype": "video",
		"name": "Tutorial: A Detailed Look at React Native for JavaScript Developers\n",
		"end_date": "2119-04-19T14:54:23.623000+00:00",
		"is_enabled": 1,
		"nocache": 0,
		"is_active": true,
		"uri": "/home/pi/screenly_assets/youtube_UExUNFNHYmxjRGUtcV9HMndXaWljWTBwazdxZFl3ajFQci41NkI0NEY2RDEwNTU3Q0M2",
		"duration": "3791",
		"play_order": 1,
		"start_date": "2018-04-19T14:54:23.623000+00:00",
		"is_processing": 0
	},
	{
		"asset_id": "youtube_UExUNFNHYmxjRGUtcV9HMndXaWljWTBwazdxZFl3ajFQci4yODlGNEE0NkRGMEEzMEQy",
		"mimetype": "video",
		"name": "Production Progressive Web Apps With JavaScript Frameworks (Google I/O '17)\n",
		"end_date": "2119-04-19T14:54:23.623000+00:00",
		"is_enabled": 1,
		"nocache": 0,
		"is_active": true,
		"uri": "/home/pi/screenly_assets/youtube_UExUNFNHYmxjRGUtcV9HMndXaWljWTBwazdxZFl3ajFQci4yODlGNEE0NkRGMEEzMEQy",
		"duration": "2432",
		"play_order": 1,
		"start_date": "2018-04-19T14:54:23.623000+00:00",
		"is_processing": 0
	}
]
```
Any Idea, why this happens

antonmolodykh, 2018-05-02:
@xstable It seems the problem really is. I was able to reproduce a similar behavior when I sent the data in the following format:
`ids: "id,next_id,other_id"`
This is a mistake in the documentation. Here's the right version:
`ids: id,next_id,other_id` (without quotes)
please try this and let me know if there are any problems.