Screenly/Anthias #268: External Data Source for Assets

srocke, 2015-02-26:
My client can't use the Asset Management or monitoring pieces offered in Pro. I need to reference content from an external database to be shown on Screenly OSE. Any ideas as to how to get this set up?

vpetersson, 2015-02-26:
The only way I can think of doing that right now is using some sort of cronjob that reads in the data source and populates the local database. 

The database on disk is a pretty simple sqlite database, so it should be pretty trivial to interact with it using your language of choice.

srocke, 2015-02-27:
@vpetersson I now have a script running in data with curl and pushing into sqlite db. One question I had was how I can gracefully start over even if the current asset is playing... thoughts?

vpetersson, 2015-02-27:
@srocke While note graceful per se, the best approach I can think of is simply `pkill -f viewer.py`.

kpdillon, 2015-03-25:
@srocke - any chance you could share your script?

srocke, 2015-03-25:
@kpdillon @vpetersson I just need to fish out an image and then I'll post the piece that runs it...

The basic run down is: CURL to PHP page on your server, loop through what you wish to do on the Pi and then eval the raw response back to run it on the command line. Make sense?

kpdillon, 2015-03-25:
Thank you.  Your explanation makes perfect sense.  Unfortunately I don't
have enough coding experience to implement, so I really appreciate your
help!

vpetersson, 2015-09-23:
Closing. Maybe we'll open up and document the API more going forward, which will be a far cleaner solution.

closed 2015-09-23 by vpetersson