Screenly/Anthias #270: Boot Splash Screen

Josselin23, 2015-03-04:
Hi,

I think it will be a good thing to add a boot splash screen to hide the "boot-lines", with Plymouth or something other.

Please tell me if it's possible.

THX

vpetersson, 2015-03-04:
We already do this with Screenly Pro (and maybe even the official OSE builds). It's fairly trivial to do with `fbi`. Not sure if it makes sense to make part of the installation script tho.

Details can be found [here](http://www.edv-huber.com/index.php/problemloesungen/15-custom-splash-screen-for-raspberry-pi-raspbian).

closed 2015-03-04 by vpetersson

Josselin23, 2015-03-04:
Thanks for the quick answer. I've already tried to do it through fbi, but it's a dirty solution.
What are you using in the Screenly Pro ?

vpetersson, 2015-03-04:
We're using `fbi` for Screenly Pro and it works fine.

Josselin23, 2015-03-04:
Ok thanks. I would like to have some informations about what happen behind the splash, like "configuring network interfaces..." or "checking updates..." or a progress bar.

I've already seen it on other distrib, and it's so pretty and clean. 

Thanks for the answer.