Screenly/Anthias #271: Handling uzbl crash when showing only 1 (webpage) asset

WimRoose, 2015-03-20:
I have a screenly ose running just one webpage that loads a new image every 30 sec. Sometimes uzbl crashes (due to javascript load?), and isn't automatically reloaded by viewer.py, since it only does a check when it has to show another asset in a playlist. Temporary fix is showing 2 assets with the same page.

vpetersson, 2015-03-20:
Hmm, could you post some debug logs? (/var/log/syslog and /tmp/screenly/*.log)

WimRoose, 2015-03-24:
@vpetersson, with a slight delay:

> syslog:
```
[110514.138255] Out of memory: Kill process 17533 (uzbl-core) score 924 or sacrifice child
[110514.138269] Killed process 17533 (uzbl-core) total-vm:1688644kB, anon-rss:704172kB, file-rss:157888kB
```
> screenly_viewer:
```
Tue, 24 Mar 2015 10:20:31 Showing asset FestivalFootage (webpage)
Tue, 24 Mar 2015 10:20:31 Sleeping for 300
Tue, 24 Mar 2015 10:25:32 Showing asset FestivalFootage2 (webpage)
Tue, 24 Mar 2015 10:25:32 Sleeping for 300
```
> tail xloader.log
```
EVENT [2972] VARIABLE_SET status_format str '<b>@[@TITLE]@</b> - @[@uri]@ - <span foreground="#bbb">@NAME</span>'
EVENT [2972] VARIA... (696779 more, please see e.stdout)
STDERR:
(uzbl-core:2972): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-label-select-on-focus after class was initialised
(uzbl-core:2972): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-scrolled-window-placement after class was initialised
Cleaning up...
``` 

So uzbl is killed because Pi is running out of memory, but isn't respawned by screenly. "FestivalFootage2" and "FestivalFootage" are actually the same, trying to force screenly reloading uzbl, but it doesn't

vpetersson, 2015-03-24:
Ok, yeah, you're most likely need to make those pages more light weight. They're too heavy. Alternatively, you might want to try out the new Raspberry Pi 2.

WimRoose, 2015-03-25:
It's already on a Pi 2, and the only javascript code I'm using is a jQuery request every 30s to load a new (random) image. I'll take a closer look at that one. But besides that, the issue is about the fact that viewer.py doesn't check if uzbl is still alive when showing just one single asset:

```
if url == current_browser_url and not force:
        logging.debug('Already showing %s, keeping it.', current_browser_url)
    else:
        current_browser_url = url
        browser_send('uri ' + current_browser_url, cb=cb)
        logging.info('current url is %s', current_browser_url)
```
Quick fix is adding a force in asset loop, like this:
```
elif 'web' in mime:
            browser_url(uri, force=True)
```
Shouldn't there be a independent function to check if uzbl is still alive so we can 
> if url == current_browser_url and not force and uzbl_alive = True:

I'll see if I can propose something like that - or am I missing something?

vpetersson, 2015-03-25:
Yeah, that might be something we could look into down the road. However, it is discouraged to have long-running web assets with dynamic content. The node will likely run out of memory (just like you're experiencing). 

A better approach would be to drop the jQuery reload and instead set the duration of the page to 30 seconds and let the viewer handle the reload. 

On a desktop box, jQuery is pretty lightweight, but on the Raspberry Pi, jQuery is likely enough to push the device to the limit

WimRoose, 2015-03-25:
Okay; thanks voor the tip

closed 2015-03-25 by WimRoose

jameskirsop, 2015-09-03:
@WimRoose, just as a tip - I've rolled my own Uzbl install (downloaded the source from https://github.com/uzbl/uzbl and then `make` and `make install` after you've got the dependencies installed).
I've found this, in my experience, to be much less hungry on memory when using the latest dependencies and build of Uzbl. I'm down from a crash roughly every hour or two, to a crash maybe every four hours or so.