Screenly/Anthias #272: Unable to add https:// assets

moodytractor, 2015-03-24:
We have tried twice now to get Screenly to display Wordpress blogs (https secure webpages) and have been unable to do so. Today we ran the latest Screenly update before trying again.

We installed the latest update then ran the command to enable SSL. Like last time, all this has resulted in is a blank screen and an inability to access the machine through either the :8080 ip address historically used to access the screenly OSE  or the :443 ip address suggested for using Screenly after running the SSL command.

Please advise how we might fix this problem. We only use the system for one screen and as a small junior school, we cannot afford the Pro edition.

vpetersson, 2015-03-24:
There is some confusion here. SSL support for the server (i.e. how you access the management interface) with SSL in the viewer component. 

In any case, it appears as something went wrong with the installation then. Please post the logs from `/tmp/screenly*` as well as the output when running the command.

moodytractor, 2015-03-24:
We don't have the miniSD adapter. Is there any way of accessing the log on the terminal itself? A command that can be written?

vpetersson, 2015-03-24:
Yes, you can access the virtual terminal. However, if you're not familiar with Linux, I recommend you simply re-flash the SD card and merely run the upgrade (and not the SSL component).

moodytractor, 2015-03-24:
[image: image1.JPG]

moodytractor, 2015-03-24:
[image]

moodytractor, 2015-03-24:
[image]

over64, 2015-11-26:
tested https://github.com/wireload/screenly-ose/commit/0c6fec271f05dfe3874bfed25ef39453372e1907
on yandex.ru, google.com, twitter.com
works fine without any special settings

vpetersson, 2015-11-26:
Hmm could maybe be an NTP related issues? In any case. Closing out this as unreproduceable then.

closed 2015-11-26 by vpetersson