Screenly/Anthias #273: Alert when new Software Update is available?

danbsfc, 2015-03-25:
I am using Screenly OSE on 2 monitors at work and they are just perfect for what we need! One thing that would be nice is to somehow be informed via the UI 'Web' Interface whenever a new software/firmware update is available to download and install?

Is this possible? Also could it be executed from the browser or would we need to still do it via SSH?

Thanks

vpetersson, 2015-03-25:
We already display that in the web interface. You will however need to run the upgrade over SSH.

closed 2015-03-25 by vpetersson

anaglyph, 2015-03-25:
It already does inform you of an 'Update Available'.

[image]

Automatic security updates and Screenly's central software management is available on the paid version, so is possible. http://www.screenlyapp.com/

danbsfc, 2015-03-25:
Strange, i've never seen / had that?