Screenly/Anthias #274: Image Size in html_templates.py

kpdillon, 2015-03-25:
I could not get images to display full screen on a standard HD TV display.  I changed my code form this

document.body.style.backgroundSize = i.width > window.innerWidth || i.height > window.innerHeight ? 'contain' : 'auto';

to 

document.body.style.backgroundSize = 'contain';

And now the images show up full screen perfectly.  I am not a developer so I can't really understand the purpose of the original code.  Wanted to share in case this is some kind of bug.  It seems to be for me.

over64, 2015-10-05:
Original code will display images which size less then screen size without scale. In other case it will scale down . Presented fix will scale up image in this situation. Is scale up good practice for bitmap images? Scale up is good for svg(vector) but with bitmap it can produce low quality result. Also original code allows you to choose scale up or not. You can scale up with some image manipulation software. Bit with this fix you will have not choice. You will scale up everething.

vpetersson, 2015-10-05:
@over64 You're right. We should never scale up images (only down). 

I wasn't sure exactly what this code did, so I wanted to investigate it.

Rejecting issue then.

closed 2015-10-05 by vpetersson