Screenly/Anthias #275: Test 'viewer_test.TestEmptyPl.test_empty' fails

vpetersson, 2015-03-27:
```
.............E...
======================================================================
ERROR: viewer_test.TestEmptyPl.test_empty
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/pi/screenly/tests/viewer_test.py", line 39, in test_empty
    self.u.main()
  File "/home/pi/screenly/viewer.py", line 360, in main
    asset_loop(scheduler)
  File "/home/pi/screenly/viewer.py", line 288, in asset_loop
    view_image(HOME + LOAD_SCREEN)
  File "/home/pi/screenly/viewer.py", line 180, in view_image
    browser_clear()
  File "/home/pi/screenly/viewer.py", line 165, in browser_clear
    browser_url('file://' + BLACK_PAGE, force=force, cb=lambda buf: 'LOAD_FINISH' in buf and BLACK_PAGE in buf)
  File "/home/pi/screenly/viewer.py", line 175, in browser_url
    browser_send('uri ' + current_browser_url, cb=cb)
  File "/home/pi/screenly/viewer.py", line 156, in browser_send
    if cb(browser.next()):
  File "/home/pi/screenly/viewer.py", line 165, in <lambda>
    browser_url('file://' + BLACK_PAGE, force=force, cb=lambda buf: 'LOAD_FINISH' in buf and BLACK_PAGE in buf)
TypeError: argument of type 'Mock' is not iterable
-------------------- >> begin captured logging << --------------------
root: DEBUG: Reading config-file...
root: DEBUG: Could not parse setting 'viewer.resolution': No option 'resolution' in section: 'viewer'. Using default value: '1920x1080'.
root: DEBUG: Could not parse setting 'viewer.use_24_hour_clock': No option 'use_24_hour_clock' in section: 'viewer'. Using default value: 'False'.
root: DEBUG: Could not parse setting 'main.listen': No option 'listen' in section: 'main'. Using default value: '0.0.0.0:8080'.
root: DEBUG: Could not parse setting 'main.assetdir': No option 'assetdir' in section: 'main'. Using default value: 'screenly_assets'.
root: INFO: Loading browser...
root: INFO: killing previous uzbl <Mock name='m_cmd()().pid' id='24110608'>
root: INFO: Browser loading http://0.0.0.0:8080/splash_page. Running as PID <Mock name='m_cmd()().pid' id='24110608'>.
root: INFO: Playlist is empty. Sleeping for 5 seconds
--------------------- >> end captured logging << ---------------------

----------------------------------------------------------------------
Ran 17 tests in 61.492s

FAILED (errors=1)
pi@481b2df7b118:~/s
```

richard-to, 2015-04-16:
I took a look at this failing test. The issue with this particular test is that it is like an integration test. It tests the whole `main` of viewer.py. 

The problem is that we end up mocking so many functions that it's kind of a pointless test. At least it's hard to understand what it actually is testing.

If it is testing an empty playlist, shouldn't there be an simple unit test?

For this specific error, it's possible it could be solved by implementing `__iter__` on the Mock `sh.Command` object.

Here is an example of mocking `__iter__`: http://programmaticallyspeaking.com/mocking-__iter__-with-a-magic-mock-in-python.html

My suggestion would be to simplify this test. That test also takes about 60 seconds due to this line `sleep(SPLASH_DELAY)`. We could mock SPLASH_DELAY, but it gets to the point where we're needing to mock way too many things and we end up testing nothing.

vpetersson, 2015-04-20:
Sorry for the delay, @richard-to (been on the road). @aljungberg will give you some feedback on this soon.

aljungberg, 2015-04-20:
@richard-to Sometimes a few integration-y tests can go a long way, especially when mixed with more specific tests. In this case though you're right, the mocking here is messy. viewer.py doesn't lend itself to easy testing since it uses globals and isn't broken down into particularly decoupled methods.

We can easily get rid of the long delay, at least, by changing viewer.SPLASH_DELAY to 0 in the test case setup method. I pushed that. I wouldn't say that this reduces the validity of the test, in my experience changing settings for testing purposes is common and perfectly okay.

As for the error in the log though, that's a little trickier. Ideally we'd simplify a little here. Especially the way we mock the browser is awkward since we have to change what sh.Command returns, rather than some easy to identify "spawn_browser" function. Might be easier to change viewer.py slightly to be a little less gnarly.

closed 2018-08-13 by vpetersson