Screenly/Anthias #277: Upgrade Fails

Cyberprog, 2015-03-30:
Hi All,

Getting the following when running the upgrade script;

pi@cf-screenly0 ~ $ ~/screenly/misc/run_upgrade.sh
bash: line 2: syntax error near unexpected token `newline'
bash: line 2: `<?xml version="1.0" encoding="utf-8"?>'

I'm guessing something wrong at the github end?

Cheers!

Alex.

vpetersson, 2015-03-30:
I just tried and it works for me.

This is most likely caused by the ongoing [DDoS attack](https://status.github.com/) on Github.

closed 2015-03-30 by vpetersson

kpdillon, 2015-03-30:
I have run the upgrade an I still see a message that says Update available in the GUI.  Is that supposed to be there?  Also, where can I look to see what bugs were fixed?

vpetersson, 2015-03-31:
This could actually also be related to the DDoS attack, since it depends on checking the commit hashes from Github.

That said, it does seem to work on the [demo site](http://ose.demo.screenlyapp.com/).

Cyberprog, 2015-03-31:
I was able to get the update run this morning - so I wonder if it hit a cloudflare page or something similar that caused it to b0rk yesterday.

vpetersson, 2015-03-31:
To my knowledge, Github isn't using Cloudflare, so that shouldn't interfere. In any case, glad you got it working.

Cyberprog, 2015-03-31:
May have been something similar :)