Screenly/Anthias #278: Images not working

rsenio, 2015-04-01:
Just updated screenly as I noticed there was an update available message on the web gui. After doing so, and a reboot, none of my images (jpg) are displayed anymore. I loaded up a new sd card using the pre-built image from the OSE site date Feb 25th 2015. Sample screen displays the comparisons etc. Once I delete those and add one of my jpg assets they do not display once again

rsenio, 2015-04-01:
Thought this may have been ntp related. But after verifying the Pi is the correct time and timezone my images do not display.

"system info" shows

Wed, 01 Apr 2015 13:32:40 Showing asset Welcome (image)
Wed, 01 Apr 2015 13:32:41 Sleeping for 5
Wed, 01 Apr 2015 13:32:46 Showing asset Welcome (image)
Wed, 01 Apr 2015 13:32:47 Sleeping for 5
Wed, 01 Apr 2015 13:32:53 Showing asset Welcome (image)
Wed, 01 Apr 2015 13:32:54 Sleeping for 5
Wed, 01 Apr 2015 13:32:59 Showing asset Welcome (image)
Wed, 01 Apr 2015 13:33:00 Sleeping for 5
Wed, 01 Apr 2015 13:33:05 Showing asset Welcome (image)

This is the same image I've been using since day 1, and it does not display anymore. No jpgs are.
If I move the jpg to a web serer and use the URL, it displays fine.

cabsports, 2015-04-02:
I am also experiencing this issue.  I am using a Raspberry Pi B+

vpetersson, 2015-04-02:
@rsenio can you post the logs from /tmp/screenly/*.log?

aaronshepler, 2015-04-02:
We are having a similar issue after updating.  I cannot view more than one image asset, as soon as a second is added they don't show or transition.

I am able to use webpages, they seem to transition fine.  While experimenting I found if I put a webpage in between each image everything works ok

rsenio, 2015-04-02:
No such directory. I have /tmp/screenly_viewer.log, sreenly_xloader,log and screenly_watchdog.log but no /tmp/screenly/

vpetersson, 2015-04-02:
@rsenio Ah ok, yeah i meant `/tmp/screely*.log`

rsenio, 2015-04-02:
Anyway, I took one of my Screenly's that had the update done on it yesteday and turned off the existing assets (video and website). I then uploaded a jpg file as a new asset. It does not "play". Here is  the contents of screenly_viewer.log as of that change this morning

rsenio, 2015-04-02:
Thu, 02 Apr 2015 08:01:58 Showing asset Client Presentation (video)
Thu, 02 Apr 2015 08:02:00 current url is file:///tmp/screenly_html/black_page.html
Thu, 02 Apr 2015 08:04:26 Showing asset mPanel (webpage)
Thu, 02 Apr 2015 08:04:27 current url is http://designelemental.net/mpanel/index.php?p=af1f8a3862d57b3e1f9825aebbfe4935
Thu, 02 Apr 2015 08:04:27 Sleeping for 600
Thu, 02 Apr 2015 08:14:27 Generating asset-list...
Thu, 02 Apr 2015 08:14:27 Playlist is empty. Sleeping for 5 seconds
Thu, 02 Apr 2015 08:14:28 current url is file:///tmp/screenly_html/black_page.html
Thu, 02 Apr 2015 08:14:34 Playlist is empty. Sleeping for 5 seconds
Thu, 02 Apr 2015 08:14:40 Playlist is empty. Sleeping for 5 seconds
Thu, 02 Apr 2015 08:14:47 Playlist is empty. Sleeping for 5 seconds
Thu, 02 Apr 2015 08:14:59 Generating asset-list...
Thu, 02 Apr 2015 08:14:59 Showing asset Test (image)
Thu, 02 Apr 2015 08:14:59 Sleeping for 10
Thu, 02 Apr 2015 08:15:10 Showing asset Test (image)
Thu, 02 Apr 2015 08:15:11 Sleeping for 10
Thu, 02 Apr 2015 08:15:21 Showing asset Test (image)
Thu, 02 Apr 2015 08:15:22 Sleeping for 10
Thu, 02 Apr 2015 08:15:33 Showing asset Test (image)
Thu, 02 Apr 2015 08:15:34 Sleeping for 10
Thu, 02 Apr 2015 08:15:45 Showing asset Test (image)
Thu, 02 Apr 2015 08:15:46 Sleeping for 10
Thu, 02 Apr 2015 08:15:58 Showing asset Test (image)
Thu, 02 Apr 2015 08:15:59 Sleeping for 10
Thu, 02 Apr 2015 08:16:09 Showing asset Test (image)
Thu, 02 Apr 2015 08:16:10 Sleeping for 10
Thu, 02 Apr 2015 08:16:21 Showing asset Test (image)
Thu, 02 Apr 2015 08:16:22 Sleeping for 10
Thu, 02 Apr 2015 08:16:32 Showing asset Test (image)
Thu, 02 Apr 2015 08:16:33 Sleeping for 10
Thu, 02 Apr 2015 08:16:43 Showing asset Test (image)
Thu, 02 Apr 2015 08:16:44 Sleeping for 10
Thu, 02 Apr 2015 08:16:55 Showing asset Test (image)
Thu, 02 Apr 2015 08:16:56 Sleeping for 10
Thu, 02 Apr 2015 08:17:06 Showing asset Test (image)
Thu, 02 Apr 2015 08:17:07 Sleeping for 10
Thu, 02 Apr 2015 08:17:18 Showing asset Test (image)
Thu, 02 Apr 2015 08:17:19 Sleeping for 10
Thu, 02 Apr 2015 08:17:30 Showing asset Test (image)
Thu, 02 Apr 2015 08:17:31 Sleeping for 10
Thu, 02 Apr 2015 08:17:41 Showing asset Test (image)
Thu, 02 Apr 2015 08:17:42 Sleeping for 10
Thu, 02 Apr 2015 08:17:52 Showing asset Test (image)
Thu, 02 Apr 2015 08:17:53 Sleeping for 10
Thu, 02 Apr 2015 08:18:04 Showing asset Test (image)
Thu, 02 Apr 2015 08:18:05 Sleeping for 10
Thu, 02 Apr 2015 08:18:16 Showing asset Test (image)
Thu, 02 Apr 2015 08:18:16 Sleeping for 10
Thu, 02 Apr 2015 08:18:28 Showing asset Test (image)
Thu, 02 Apr 2015 08:18:28 Sleeping for 10
Thu, 02 Apr 2015 08:18:39 Showing asset Test (image)
Thu, 02 Apr 2015 08:18:40 Sleeping for 10
Thu, 02 Apr 2015 08:18:53 Showing asset Test (image)
Thu, 02 Apr 2015 08:18:54 Sleeping for 10
Thu, 02 Apr 2015 08:19:07 Showing asset Test (image)
Thu, 02 Apr 2015 08:19:08 Sleeping for 10
Thu, 02 Apr 2015 08:19:18 Showing asset Test (image)
Thu, 02 Apr 2015 08:19:19 Sleeping for 10
Thu, 02 Apr 2015 08:19:37 Showing asset Test (image)
Thu, 02 Apr 2015 08:19:38 Sleeping for 10
Thu, 02 Apr 2015 08:19:48 Showing asset Test (image)
Thu, 02 Apr 2015 08:19:49 Sleeping for 10
Thu, 02 Apr 2015 08:20:01 Showing asset Test (image)
Thu, 02 Apr 2015 08:20:01 Sleeping for 10
Thu, 02 Apr 2015 08:20:15 Showing asset Test (image)
Thu, 02 Apr 2015 08:20:16 Sleeping for 10
Thu, 02 Apr 2015 08:20:30 Showing asset Test (image)
Thu, 02 Apr 2015 08:20:31 Sleeping for 10
Thu, 02 Apr 2015 08:20:59 Showing asset Test (image)
Thu, 02 Apr 2015 08:21:00 Sleeping for 10
Thu, 02 Apr 2015 08:21:19 Showing asset Test (image)
Thu, 02 Apr 2015 08:21:20 Sleeping for 10
Thu, 02 Apr 2015 08:21:36 Showing asset Test (image)
Thu, 02 Apr 2015 08:21:37 Sleeping for 10
Thu, 02 Apr 2015 08:21:47 Showing asset Test (image)
Thu, 02 Apr 2015 08:21:48 Sleeping for 10
Thu, 02 Apr 2015 08:21:59 Showing asset Test (image)
Thu, 02 Apr 2015 08:21:59 Sleeping for 10
Thu, 02 Apr 2015 08:22:11 Showing asset Test (image)
Thu, 02 Apr 2015 08:22:11 Sleeping for 10
Thu, 02 Apr 2015 08:22:22 Showing asset Test (image)
Thu, 02 Apr 2015 08:22:23 Sleeping for 10
Thu, 02 Apr 2015 08:22:34 Showing asset Test (image)
Thu, 02 Apr 2015 08:22:34 Sleeping for 10
Thu, 02 Apr 2015 08:22:45 Showing asset Test (image)
Thu, 02 Apr 2015 08:22:46 Sleeping for 10
Thu, 02 Apr 2015 08:22:56 Showing asset Test (image)
Thu, 02 Apr 2015 08:22:57 Sleeping for 10
Thu, 02 Apr 2015 08:23:08 Showing asset Test (image)

vpetersson, 2015-04-02:
Hmm, nothing interesting there.

Could you try turning on debug logging. Change 'debug_logging = False' to 'debug_logging = True'  in `~/.screenly/screenly.conf`

rsenio, 2015-04-02:
I cleared the log, enabled the debug option and rebooted with only a jpg asset turned on. Here is the log

Thu, 02 Apr 2015 08:33:19 Loading browser...
Thu, 02 Apr 2015 08:33:19 Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 2590.
Thu, 02 Apr 2015 08:33:24 Scheduler init
Thu, 02 Apr 2015 08:33:24 update_playlist
Thu, 02 Apr 2015 08:33:24 Generating asset-list...
Thu, 02 Apr 2015 08:33:24 generate_asset_list deadline: 2015-04-09 14:14:00
Thu, 02 Apr 2015 08:33:24 update_playlist done, count 1, counter 0, index 0, deadline 2015-04-09 14:14:00
Thu, 02 Apr 2015 08:33:24 Entering infinite loop.
Thu, 02 Apr 2015 08:33:24 Last update: 2015-03-24 12:53:11.874985
Thu, 02 Apr 2015 08:33:26 Received non 200-status
Thu, 02 Apr 2015 08:33:26 get_next_asset
Thu, 02 Apr 2015 08:33:26 refresh_playlist
Thu, 02 Apr 2015 08:33:26 refresh: counter: (0) deadline (2015-04-09 14:14:00) timecur (2015-04-02 14:33:26.305581)
Thu, 02 Apr 2015 08:33:26 get_next_asset after refresh
Thu, 02 Apr 2015 08:33:26 get_next_asset counter 0 returning asset 1 of 1
Thu, 02 Apr 2015 08:33:26 Showing asset Test (image)
Thu, 02 Apr 2015 08:33:26 Asset URI /home/pi/screenly_assets/48827eda08f1453cbd857866c03b2793
Thu, 02 Apr 2015 08:33:26 Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Thu, 02 Apr 2015 08:33:28 Sleeping for 3
Thu, 02 Apr 2015 08:33:31 Last update: 2015-03-24 12:53:11.874985
Thu, 02 Apr 2015 08:33:31 Received non 200-status
Thu, 02 Apr 2015 08:33:31 get_next_asset
Thu, 02 Apr 2015 08:33:31 refresh_playlist
Thu, 02 Apr 2015 08:33:31 refresh: counter: (0) deadline (2015-04-09 14:14:00) timecur (2015-04-02 14:33:31.746948)
Thu, 02 Apr 2015 08:33:31 get_next_asset after refresh
Thu, 02 Apr 2015 08:33:31 get_next_asset counter 0 returning asset 1 of 1
Thu, 02 Apr 2015 08:33:31 Showing asset Test (image)
Thu, 02 Apr 2015 08:33:31 Asset URI /home/pi/screenly_assets/48827eda08f1453cbd857866c03b2793
Thu, 02 Apr 2015 08:33:31 Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Thu, 02 Apr 2015 08:33:32 Sleeping for 3
Thu, 02 Apr 2015 08:33:35 Last update: 2015-03-24 12:53:11.874985
Thu, 02 Apr 2015 08:33:36 Received non 200-status
Thu, 02 Apr 2015 08:33:36 get_next_asset
Thu, 02 Apr 2015 08:33:36 refresh_playlist
Thu, 02 Apr 2015 08:33:36 refresh: counter: (0) deadline (2015-04-09 14:14:00) timecur (2015-04-02 14:33:36.155564)
Thu, 02 Apr 2015 08:33:36 get_next_asset after refresh
Thu, 02 Apr 2015 08:33:36 get_next_asset counter 0 returning asset 1 of 1
Thu, 02 Apr 2015 08:33:36 Showing asset Test (image)
Thu, 02 Apr 2015 08:33:36 Asset URI /home/pi/screenly_assets/48827eda08f1453cbd857866c03b2793
Thu, 02 Apr 2015 08:33:36 Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Thu, 02 Apr 2015 08:33:37 Sleeping for 3
Thu, 02 Apr 2015 08:33:40 Last update: 2015-03-24 12:53:11.874985
Thu, 02 Apr 2015 08:33:40 Received non 200-status
Thu, 02 Apr 2015 08:33:40 get_next_asset
Thu, 02 Apr 2015 08:33:40 refresh_playlist
Thu, 02 Apr 2015 08:33:40 refresh: counter: (0) deadline (2015-04-09 14:14:00) timecur (2015-04-02 14:33:40.535934)
Thu, 02 Apr 2015 08:33:40 get_next_asset after refresh
Thu, 02 Apr 2015 08:33:40 get_next_asset counter 0 returning asset 1 of 1
Thu, 02 Apr 2015 08:33:40 Showing asset Test (image)
Thu, 02 Apr 2015 08:33:40 Asset URI /home/pi/screenly_assets/48827eda08f1453cbd857866c03b2793
Thu, 02 Apr 2015 08:33:40 Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Thu, 02 Apr 2015 08:33:41 Sleeping for 3
Thu, 02 Apr 2015 08:33:44 Last update: 2015-03-24 12:53:11.874985
Thu, 02 Apr 2015 08:33:44 Received non 200-status
Thu, 02 Apr 2015 08:33:44 get_next_asset
Thu, 02 Apr 2015 08:33:44 refresh_playlist
Thu, 02 Apr 2015 08:33:44 refresh: counter: (0) deadline (2015-04-09 14:14:00) timecur (2015-04-02 14:33:44.957924)
Thu, 02 Apr 2015 08:33:44 get_next_asset after refresh
Thu, 02 Apr 2015 08:33:44 get_next_asset counter 0 returning asset 1 of 1
Thu, 02 Apr 2015 08:33:44 Showing asset Test (image)
Thu, 02 Apr 2015 08:33:44 Asset URI /home/pi/screenly_assets/48827eda08f1453cbd857866c03b2793
Thu, 02 Apr 2015 08:33:44 Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Thu, 02 Apr 2015 08:33:45 Sleeping for 3
Thu, 02 Apr 2015 08:33:48 Last update: 2015-03-24 12:53:11.874985
Thu, 02 Apr 2015 08:33:49 Received non 200-status
Thu, 02 Apr 2015 08:33:49 get_next_asset
Thu, 02 Apr 2015 08:33:49 refresh_playlist
Thu, 02 Apr 2015 08:33:49 refresh: counter: (0) deadline (2015-04-09 14:14:00) timecur (2015-04-02 14:33:49.340741)
[... 91 lines cut ...]
Thu, 02 Apr 2015 08:34:24 Asset URI /home/pi/screenly_assets/48827eda08f1453cbd857866c03b2793
Thu, 02 Apr 2015 08:34:24 Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Thu, 02 Apr 2015 08:34:25 Sleeping for 3
Thu, 02 Apr 2015 08:34:28 Last update: 2015-03-24 12:53:11.874985
Thu, 02 Apr 2015 08:34:28 Received non 200-status
Thu, 02 Apr 2015 08:34:28 get_next_asset
Thu, 02 Apr 2015 08:34:28 refresh_playlist
Thu, 02 Apr 2015 08:34:28 refresh: counter: (0) deadline (2015-04-09 14:14:00) timecur (2015-04-02 14:34:28.911921)
Thu, 02 Apr 2015 08:34:28 get_next_asset after refresh
Thu, 02 Apr 2015 08:34:28 get_next_asset counter 0 returning asset 1 of 1
Thu, 02 Apr 2015 08:34:28 Showing asset Test (image)
Thu, 02 Apr 2015 08:34:28 Asset URI /home/pi/screenly_assets/48827eda08f1453cbd857866c03b2793
[... cut ...]
Thu, 02 Apr 2015 08:34:33 refresh: counter: (0) deadline (2015-04-09 14:14:00) timecur (2015-04-02 14:34:33.291893)
Thu, 02 Apr 2015 08:34:33 get_next_asset after refresh
Thu, 02 Apr 2015 08:34:33 get_next_asset counter 0 returning asset 1 of 1
Thu, 02 Apr 2015 08:34:33 Showing asset Test (image)
Thu, 02 Apr 2015 08:34:33 Asset URI /home/pi/screenly_assets/48827eda08f1453cbd857866c03b2793
Thu, 02 Apr 2015 08:34:33 Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Thu, 02 Apr 2015 08:34:34 Sleeping for 3

vpetersson, 2015-04-02:
This part is interesting:

     [...] Received non 200-status

This means that the player tried to fetch a web asset and received a non-clean status from the server.

rsenio, 2015-04-02:
Only asset to play is an uploaded jpg

vpetersson, 2015-04-02:
Hmm ok. I don't have any player here right now, so I cannot test it.

aaronshepler, 2015-04-02:
FWIW, I am showing that same non 200-status on all 8 of mine

Thu, 02 Apr 2015 09:53:44 Sleeping for 90
Thu, 02 Apr 2015 09:54:10 Loading browser...
Thu, 02 Apr 2015 09:54:10 Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 7991.
Thu, 02 Apr 2015 09:54:14 Scheduler init
Thu, 02 Apr 2015 09:54:14 update_playlist
Thu, 02 Apr 2015 09:54:14 Generating asset-list...
Thu, 02 Apr 2015 09:54:14 generate_asset_list deadline: 2015-04-15 12:53:00
Thu, 02 Apr 2015 09:54:14 update_playlist done, count 2, counter 0, index 0, deadline 2015-04-15 12:53:00
Thu, 02 Apr 2015 09:54:14 Entering infinite loop.
Thu, 02 Apr 2015 09:54:14 Last update: 2015-03-24 09:42:51.021701
Thu, 02 Apr 2015 09:54:17 Received non 200-status
Thu, 02 Apr 2015 09:54:17 get_next_asset
Thu, 02 Apr 2015 09:54:17 refresh_playlist
Thu, 02 Apr 2015 09:54:17 refresh: counter: (0) deadline (2015-04-15 12:53:00) timecur (2015-04-02 14:54:17.063154)
Thu, 02 Apr 2015 09:54:17 get_next_asset after refresh
Thu, 02 Apr 2015 09:54:17 get_next_asset counter 0 returning asset 1 of 2
Thu, 02 Apr 2015 09:54:17 Showing asset 25 march news page (image)
Thu, 02 Apr 2015 09:54:17 Asset URI /home/pi/screenly_assets/65445f7d64104e0f9697d1c0c5597197
Thu, 02 Apr 2015 09:54:17 Already showing file:///tmp/screenly_html/black_page.html, reloading it.
Thu, 02 Apr 2015 09:54:18 Sleeping for 90

rsenio, 2015-04-02:
Downloaded the image from the site, applied it to a new SD card. Booted up, deleted the sample assest and uploaded a jpg. Didn't adjust the system time at all, but set the asset to start way in the past. Worked fine. Ran  ~/screenly/misc/run_upgrade.sh and rebooted. Broken again

ci5ic, 2015-04-03:
Experiencing the same problems... Shows video/web assets just fine, but only shows only certain .jpg (if at all). if I re-order the asset lists, turn on/off assets, or add new assets, a different selection of .jpg assets will display while some others that were previously showing up will no longer display. Is there a way to roll back to a previous build from the command line?

thorstenhartung, 2015-04-04:
Same problem here, recognized, that after trying to update via script and everything has finished the "update available message" ist still present

richard-to, 2015-04-04:
Hmm, I'll see if I can reproduce this on my virtual machine to see if it is a software issue, which I believe it is. I haven't tested any of the recent versions of Screenly lately. I'll post back what I find out later today.

richard-to, 2015-04-04:
I was able to reproduce the bug. Test case:

1. Turn on server.py
2. Delete all existing assets
3. Add a jpg image
4. Activate image
  - Image should show up
5. Add another image
6. Activate image
  - Looks to work at first, but then I noticed the only one image shows. The other image is a black screen for 10 seconds. In my case the first uploaded image stops showing.
7. Next add a website asset and activate
  - Only the website and one image show
8. Remove website asset
  - Same behavior as 6
9. Remove the image asset that is showing
  - Original image now shows up

I haven't looked at the code yet, but it's definitely a software problem.

richard-to, 2015-04-04:
Hmm, I did notice something weird at least with my VM. I'm on Yosemite with Fusion 7, but I noticed that my second image would appear if I changed to a different fullscreen window (say my browser). When I turned back to my fullscreen VM view, I saw the second image. But then if I wait for the cycle, it showed a black screen for that second image again.

richard-to, 2015-04-05:
I found the bug. It's in `viewer.py` around line 173.

This line was recently added and is causing the problem:

    browser_send('reload full')

Once I removed the line, the images displayed correctly.

This is because `current_browser_url` equals `file:///tmp/screenly_html/black_page.html` for images, so if you have two images in a row, then the browser will reload to the black page.

This is because in `view_image`, we first do `brower_clear` which calls `browser_url`. The browser reload is performed in `browser_url`. The browser reload is asynchronous, I think, so that means there is a race condition, where the image will be loaded first, but then the browser reload occurs right after. That's just a guess though on why it does not work.

richard-to, 2015-04-05:
One possible solution it to do `browser_clear(True)` instead of `browser_clear()`. This will force a `browser_send` of the url, which you'd think was the same as a `reload full` if the url is indeed the same.

The problem looks like the `cb` parameter was not passed. For browser clear we pass the "LOAD FINISH" callback. 

For example:

    browser_send('uri ' + current_browser_url, cb=cb)

Versus:
    
    browser_send('reload full')

Maybe change it to:

    browser_send('reload full', cb=cb) 

But since both branches are now redundant, it would be almost the same as this (difference would be the logging info):

```python
def browser_url(url, cb=lambda _: True, force=False):
    global current_browser_url
    current_browser_url = url
    browser_send('uri ' + current_browser_url, cb=cb)
    logging.info('current url is %s', current_browser_url)
```
However, you definitely do not want to reload the black page for images. I noticed that I had my assets set to the default 10 seconds, but they took about 20 seconds to change. At least that's what I noticed.

Maybe a solution would be to check if the black page is set in `view_image` and let `browser_url` load each URL since that is the behavior that is desired for webpages. 

Pseudo code example:

```python
def view_image(uri):
    if not_black_page():
        browser_clear()
    browser_send('js window.setimg("{0}")'.format(uri), cb=lambda b: 'COMMAND_EXECUTED' in b and 'setimg' in b)
```

vpetersson, 2015-04-06:
Sorry guys, this is my bad. I pushed that fix recklessly. We really need a test here. 

@richard-to from what I can tell, this is the proper solution:

    browser_send('uri ' + current_browser_url, cb=cb)

richard-to, 2015-04-06:
@vpetersson 

Yeah, I agree about needing a test here. A few of the viewer.py unit tests I think are broken as well, so it may be a good time to fix those too.

I think the use cases are the following:

1. Given a single asset with the URL  http://www.screenlyapp.com:
  - Screenly should load http://www.screenlyapp.com after each duration
2. Given two assets with the URL http://www.screenlyapp.com:
  - Screenly should load http://www.screenlyapp.com after each duration
3. Given two different webpage assets:
  - Screenly should load each webpage after each duration
4. Given three different image assets:
  - Screenly should load the black/blank page if that page is not currently showing
  - Screenly should **not** reload the blank/black page URL when displaying consecutive images

For the first three use cases, the following would work since Screenly will need to load the webpage in every use case involving webpages.

```python
def browser_url(url, cb=lambda _: True, force=False):
    global current_browser_url
    current_browser_url = url
    browser_send('uri ' + current_browser_url, cb=cb)
    logging.info('current url is %s', current_browser_url)
```

The problem is that you do not want to reload the black/blank webpage each time when displaying images. I could be wrong, but it seemed to delay the loading of the next image. As mentioned in the previous message, the image took 20 seconds to load instead of 10 seconds.

I'll have to double check that later. If that turns out not to be the case then the above snippet should work fine. Otherwise, you'd probably need something like this:

```python
def view_image(uri):
    if not_black_page():
        browser_clear()
    browser_send('js window.setimg("{0}")'.format(uri), cb=lambda b: 'COMMAND_EXECUTED' in b and 'setimg' in b)
```

vpetersson, 2015-04-06:
@richard-to Yes, looks like some tests are failing:

     .............E...
     ======================================================================
     ERROR: viewer_test.TestEmptyPl.test_empty
     ----------------------------------------------------------------------
     Traceback (most recent call last):
       File "/usr/local/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
         self.test(*self.arg)
       File "/home/pi/screenly/tests/viewer_test.py", line 39, in test_empty
         self.u.main()
       File "/home/pi/screenly/viewer.py", line 361, in main
         asset_loop(scheduler)
       File "/home/pi/screenly/viewer.py", line 289, in asset_loop
         view_image(HOME + LOAD_SCREEN)
       File "/home/pi/screenly/viewer.py", line 181, in view_image
         browser_clear()
       File "/home/pi/screenly/viewer.py", line 165, in browser_clear
         browser_url('file://' + BLACK_PAGE, force=force, cb=lambda buf: 'LOAD_FINISH' in buf and BLACK_PAGE in buf)
       File "/home/pi/screenly/viewer.py", line 176, in browser_url
         browser_send('uri ' + current_browser_url, cb=cb)
       File "/home/pi/screenly/viewer.py", line 156, in browser_send
         if cb(browser.next()):
       File "/home/pi/screenly/viewer.py", line 165, in <lambda>
         browser_url('file://' + BLACK_PAGE, force=force, cb=lambda buf: 'LOAD_FINISH' in buf and BLACK_PAGE in buf)
     TypeError: argument of type 'Mock' is not iterable
     -------------------- >> begin captured logging << --------------------
     root: DEBUG: Reading config-file...
     root: DEBUG: Could not parse setting 'viewer.resolution': No option 'resolution' in section: 'viewer'. Using default value: '1920x1080'.
     root: DEBUG: Could not parse setting 'viewer.use_24_hour_clock': No option 'use_24_hour_clock' in section: 'viewer'. Using default value: 'False'.
     root: DEBUG: Could not parse setting 'main.listen': No option 'listen' in section: 'main'. Using default value: '0.0.0.0:8080'.
     root: DEBUG: Could not parse setting 'main.assetdir': No option 'assetdir' in section: 'main'. Using default value: 'screenly_assets'.
     root: INFO: Loading browser...
     root: INFO: killing previous uzbl <Mock name='m_cmd()().pid' id='46193552'>
     root: INFO: Browser loading http://0.0.0.0:8080/splash_page. Running as PID <Mock name='m_cmd()().pid' id='46193552'>.
     root: INFO: Playlist is empty. Sleeping for 5 seconds
     --------------------- >> end captured logging << ---------------------
     
     ----------------------------------------------------------------------
     Ran 17 tests in 62.445s

rsenio, 2015-04-08:
So.....this is fixed...being fixed?

ci5ic, 2015-04-08:
@rsenio What I did was to manually roll back the viewer.py file on my RPi's to a previous version and it works fine at this point... Won't be running another update until this issue is resolved though.

richard-to, 2015-04-08:
I've been busy this week so haven't had time to write the unit tests and verify that the behavior is correct with my suggested patch. Hopefully in the next few days.

vpetersson, 2015-04-09:
I've rolled back this change for now.

rsenio, 2015-04-24:
Screenly is showing me another update here, if I do this am I updating to the broken version or is this taken care of

vpetersson, 2015-04-25:
No, I've reverted that change, so it should be safe.

over64, 2015-10-09:
It's alright after this https://github.com/wireload/screenly-ose/commit/fe950f641218644d14396dac060183b49e430c2c
I tested 3 images (url, local, url) and it works fine

closed 2015-10-09 by vpetersson

davezdesignz, 2016-03-30:
Check that your end date is in the future and not in the past or else they do not activate.