Screenly/Anthias #279: Getting a Black Screen

cabsports, 2015-04-21:
Hi Everyone,

I am having an issue with one of my devices where after a few hours the screen just displays a black / blank screen and I can no longer access the web interface but I can access the Raspberry Pi Unit through Putty.

The screenly_viewer.log (with debug) seem to stop being written too, and screenly_xloader.log starts filling up with lines consisting of "Cleaning up..."

Last few lines of screenly_viewer.log:

Tue, 21 Apr 2015 12:20:47 Sleeping for 7
Tue, 21 Apr 2015 12:20:54 Last update: 2014-04-26 04:47:05.915333
Tue, 21 Apr 2015 12:20:55 Received non 200-status
Tue, 21 Apr 2015 12:20:55 get_next_asset
Tue, 21 Apr 2015 12:20:55 refresh_playlist
Tue, 21 Apr 2015 12:20:55 refresh: counter: (0) deadline (2015-11-02 01:34:00) timecur (2015-04-21 02:20:55.081885)
Tue, 21 Apr 2015 12:20:55 get_next_asset after refresh
Tue, 21 Apr 2015 12:20:55 get_next_asset counter 0 returning asset 1 of 21
Tue, 21 Apr 2015 12:20:55 Showing asset Fusion Pizza Menu copy.jpg (image)
Tue, 21 Apr 2015 12:20:55 Asset URI /home/pi/screenly_assets/85571df1664a4424aaad74570722be00
Tue, 21 Apr 2015 12:20:55 Already showing file:///tmp/screenly_html/black_page.html.
Tue, 21 Apr 2015 12:20:55 Sleeping for 15
Tue, 21 Apr 2015 12:21:10 Last update: 2014-04-26 04:47:05.915333
Tue, 21 Apr 2015 12:21:10 Received non 200-status
Tue, 21 Apr 2015 12:21:10 get_next_asset
Tue, 21 Apr 2015 12:21:10 refresh_playlist
Tue, 21 Apr 2015 12:21:10 refresh: counter: (0) deadline (2015-11-02 01:34:00) timecur (2015-04-21 02:21:10.696697)
Tue, 21 Apr 2015 12:21:10 get_next_asset after refresh
Tue, 21 Apr 2015 12:21:10 get_next_asset counter 0 returning asset 2 of 21
Tue, 21 Apr 2015 12:21:10 Showing asset qld barbeque pizza.jpg (image)
Tue, 21 Apr 2015 12:21:10 Asset URI /home/pi/screenly_assets/289e2ed5664c46e5b25a24e4693f8828
Tue, 21 Apr 2015 12:21:10 Already showing file:///tmp/screenly_html/black_page.html.
Tue, 21 Apr 2015 12:21:11 Sleeping for 7
Tue, 21 Apr 2015 12:21:18 Last update: 2014-04-26 04:47:05.915333
Tue, 21 Apr 2015 12:21:18 Received non 200-status
Tue, 21 Apr 2015 12:21:18 get_next_asset
Tue, 21 Apr 2015 12:21:18 refresh_playlist
Tue, 21 Apr 2015 12:21:18 refresh: counter: (0) deadline (2015-11-02 01:34:00) timecur (2015-04-21 02:21:18.634617)
Tue, 21 Apr 2015 12:21:18 get_next_asset after refresh
Tue, 21 Apr 2015 12:21:18 get_next_asset counter 0 returning asset 3 of 21
Tue, 21 Apr 2015 12:21:18 Showing asset bbq tiger prawn.jpg (image)
Tue, 21 Apr 2015 12:21:18 Asset URI /home/pi/screenly_assets/a074a112775246f9ae97e5040d45544d
Tue, 21 Apr 2015 12:21:18 Already showing file:///tmp/screenly_html/black_page.html.
Tue, 21 Apr 2015 12:21:19 Sleeping for 7
Tue, 21 Apr 2015 12:21:26 Last update: 2014-04-26 04:47:05.915333
Tue, 21 Apr 2015 12:21:26 Received non 200-status
Tue, 21 Apr 2015 12:21:26 get_next_asset
Tue, 21 Apr 2015 12:21:26 refresh_playlist
Tue, 21 Apr 2015 12:21:26 refresh: counter: (0) deadline (2015-11-02 01:34:00) timecur (2015-04-21 02:21:26.602703)
Tue, 21 Apr 2015 12:21:26 get_next_asset after refresh
Tue, 21 Apr 2015 12:21:26 get_next_asset counter 0 returning asset 4 of 21
Tue, 21 Apr 2015 12:21:26 Showing asset hawaiian pizza.jpg (image)
Tue, 21 Apr 2015 12:21:26 Asset URI /home/pi/screenly_assets/c75906982da746399f740c91b687bb56
Tue, 21 Apr 2015 12:21:26 Already showing file:///tmp/screenly_html/black_page.html.
Tue, 21 Apr 2015 12:21:27 Sleeping for 7
Tue, 21 Apr 2015 12:21:34 Last update: 2014-04-26 04:47:05.915333
Tue, 21 Apr 2015 12:21:34 Received non 200-status
Tue, 21 Apr 2015 12:21:34 get_next_asset
Tue, 21 Apr 2015 12:21:34 refresh_playlist
Tue, 21 Apr 2015 12:21:34 refresh: counter: (0) deadline (2015-11-02 01:34:00) timecur (2015-04-21 02:21:34.656946)
Tue, 21 Apr 2015 12:21:34 get_next_asset after refresh
Tue, 21 Apr 2015 12:21:34 get_next_asset counter 0 returning asset 5 of 21
Tue, 21 Apr 2015 12:21:34 Showing asset bbq carnivore.jpg (image)
Tue, 21 Apr 2015 12:21:34 Asset URI /home/pi/screenly_assets/ffcdbe88cf4645119b5a1a22d66dd15e
Tue, 21 Apr 2015 12:21:34 Already showing file:///tmp/screenly_html/black_page.html.
Tue, 21 Apr 2015 12:21:35 Sleeping for 7
Tue, 21 Apr 2015 12:21:42 Last update: 2014-04-26 04:47:05.915333
Tue, 21 Apr 2015 12:21:42 Received non 200-status
Tue, 21 Apr 2015 12:21:42 get_next_asset

Last week I created a new Micro SD card for this instance as it was doing the same but it has made no difference.

Any ideas would be greatly received
Thanks
Duncan

vpetersson, 2015-04-21:
Hi Duncan,

Looks like something isn't working well. Maybe a simple upgrade will do the trick.

cabsports, 2015-04-22:
Hi Victor,

I did an upgrade and even tried a new Raspberry Pi unit but it still din't work.

I have loaded up another SD card and seems to be work now for about 4 hours so I will see how it is going tomorrow and report back.

vpetersson, 2015-04-22:
Sounds like a faulty SD card. I wouldn't be surprised if you find some errors in `/var/log/errors`

closed 2015-04-27 by vpetersson

cabsports, 2015-04-27:
Hi @vpetersson 

I have just had the same Raspberry Pi fail again.  It is getting a black screen, and there is no /var/log/errors file.  

This is a brand new card and lasted less than a week.  The only thing with this display is that it loops though about 20 images each one displaying for about 8 seconds.  Would this make a difference?

Thanks
Duncan

vpetersson, 2015-05-08:
@cabsports Ok, please post the logs from `/var/log/screenly*.log`. It's probably easier if you post them as [Gists](https://gist.github.com/).

cabsports, 2015-05-19:
Hi @vpetersson 

I have installed another unit in its place and Raspberry Pi 2 B and it seems to be working at the moment.

I have attached the logs as Gists and hope it works.

https://gist.github.com/cabsports/b67a4174b5fc07d12b0d
https://gist.github.com/cabsports/6457bde01dd504b3612d

vpetersson, 2015-05-19:
@cabsports Got it. I still see some strange things in the logs. I'll try to carve out some time this week to see if I can reproduce that locally.