Screenly/Anthias #280: No Video playing when network disconnected

McNeillChris9, 2015-04-26:
Hi Guys, 

This may be a miss-understanding on my part, but I was hoping that once I got the desired assets up and running that I could remove the network cable from my rPi and use screenly to just blindly play, however when I've no network cable connected I am just getting a blank (black) screen shown.

Is this something that's by-design or something that can be worked around etc? 

Thanks,
Chris

vpetersson, 2015-04-27:
Hi Chris,

I think this is still a known issue. I haven't had a chance to test this to be honest.

over64, 2015-10-09:
tested https://github.com/wireload/screenly-ose/commit/fe950f641218644d14396dac060183b49e430c2c version. Works fine

closed 2015-10-09 by vpetersson