Screenly/Anthias #283: supervisord dies when time goes backwards

brianredbeard, 2015-05-02:
Filing this issue here to track an upstream bug (Supervisor/supervisor/issues/281).  In performing a few testing scenarios around assets changing at different times I was both fast-forwarding and rewinding time.  When time goes backwards supervisord has a penchant for dying.  In this case it kills `screenly/server.py`.  This does not cause the display to fail, only to stop updating the images, i.e.:

```
pi@display0ps fauxwww | grep -2 screenly
pi        2729  0.7  1.2   6212  4656 pts/1    Ss   22:06   0:03          \_ -bash
pi        2860  0.0  0.6   4464  2188 pts/1    R+   22:15   0:00              \_ ps fauxwww
pi        2861  0.0  0.5   3556  1860 pts/1    S+   22:15   0:00              \_ grep --color=auto -2 screenly
root      2272  0.0  1.2  31816  4664 ?        Sl   22:03   0:00 /usr/sbin/lightdm
root      2302  0.1  7.0 100636 25568 tty7     Ss+  22:03   0:01  \_ /usr/bin/X -nocursor :0 -auth /var/run/lightdm/root/:0 -nolisten tcp vt7 -novtswitch
--
pi        2426  0.0  0.0   3540   216 ?        Ss   22:03   0:00          \_ /usr/bin/ssh-agent /usr/bin/dbus-launch --exit-with-session x-session-manager
pi        2453  0.0  2.5  14996  9320 ?        S    22:03   0:00          \_ openbox --config-file /home/pi/.config/openbox/lxde-rc.xml
pi        2457  0.0  0.7   4280  2568 ?        S    22:03   0:00          \_ /bin/bash /home/pi/screenly/misc/xloader.sh
pi        2490  5.5  3.6  33956 13264 ?        Sl   22:03   0:39              \_ python /home/pi/screenly/viewer.py
pi        2507  0.4  9.1 151084 33420 pts/0    Ssl+ 22:03   0:03                  \_ uzbl-core --config=- --uri=file:///tmp/screenly_html/black_page.html --print-events --connect-socket /home/pi/.cache/uzbl/event_daemon
root      2313  0.0  1.4  28648  5240 ?        Sl   22:03   0:00 /usr/sbin/console-kit-daemon --no-daemon
root      2381  0.0  1.2  23424  4492 ?        Sl   22:03   0:00 /usr/lib/policykit-1/polkitd --no-debug
```

vpetersson, 2015-05-04:
@brianredbeard Interesting scenario. Given the lack of RTC of the RasPi, I guess this is more common there than on other platforms. Not much we can do on our end, but thanks for opening this so that we can track it. Once there is a fix upstream, we can upgrade Supervisor.

mnaberez, 2015-06-25:
This was fixed in Supervisor/Supervisor@fe4edfd26ecb359580b9d13c1bd9c83ea1f07df6 and will be in the next release (3.2).  If you able to try the code on the master branch, we would appreciate if you would see if it fixes your issue.

over64, 2015-11-24:
latest version in repo (debian 7 for raspberry) is 3.0a8-1.1+deb7u1

vpetersson, 2015-11-26:
We could do like we do in Screenly Pro and install it from `pip` simply. We still need to have the init.d files, but we can extract them from the current package.

mnaberez, 2015-11-26:
> latest version in repo (debian 7 for raspberry) is 3.0a8-1.1+deb7u1

Supervisor 3.0a8 is a very old version.  It was released in 2010.

> We could do like we do in Screenly Pro and install it from pip simply. We still need to have the init.d files, but we can extract them from the current package.

If you are using such an old version, installing a newer one is a very good idea.  There have been [many bug fixes](http://supervisord.org/changes.html) since 3.0a8.

vpetersson, 2015-12-21:
Fixed in #339 and #342. Could you confirm, @brianredbeard?

closed 2015-12-21 by vpetersson