Screenly/Anthias #284: Suport to VNC and dynamic ip updates

carmolim, 2015-05-08:
Hello, I'm considering using some raspberry pi 2 and screenly ose in a client store, but I want to know if is possible to install vnc and setup a dynamic DNS. That would be great for a remote admin. 

Thanks

sequoiar, 2015-05-08:
@carmolim you can try peer-vnc on rpi2 

https://github.com/InstantWebP2P/peer-vnc

kpdillon, 2015-05-08:
Yes, this is possible although you won't be able to view the video, but
URL's and pictures will display just fine.  You can also tail the
/tmp/screenly_viewer.log to see what is displaying.  One thing I did is
setup a reverse SSH tunnel via sTunnel. This allows the PI to phone home to
me and I can get into it remotely from anywhere in the world including my
iPhone. Best part for me was the church I installed this in didn't have to
have their network admin open up any ports etc for me to get in and manage
this.  Oh I also do both RSYNC and DD backups over SSH.  Both work
fantastic!

carmolim, 2015-05-08:
@kpdillon amazing work! I don't know if I have all the knowledge to make all this configuration, but is good to know it is possible. In your experience the system is reliable? Any crash? Im worry about managing al l the screens remotely, some of the stores are in other cities.

kpdillon, 2015-05-08:
It's been rock solid for me!

vpetersson, 2015-05-08:
@carmolim You wouldn't be able to admin anything over VNC per se. All the management happens over the web interface, so an SSH tunnel for accessing the web interface would be a better approach. The only value VNC would bring would be to get a live preview of what's on the screen. Also, as @kpdillon correctly points out, you would still not be able to get the video, as that happens over frame buffer[1].

Dynamic IP is somewhat outside the scope of Screenly, but something like DynDNS should work just fine.

[1] In theory, I guess you could have two VNC sessions, one for X and one for framebuffer).

closed 2015-05-08 by vpetersson

Cyberprog, 2015-05-08:
> -----Original Message-----
> 100120817> . <https://github.com/notifications/beacon/AELTNhzer43JTZfe-

I like the idea of using VNC to view the output. That does sound like a handy feature.

pcarass, 2016-07-07:
@kpdillon ,that sounds like an amazing solution. My problem at the moment is that I'm using OSE in various different sites and everywhere has a different setup, different routers etc, and Im looking to setup something like that so i can avoid all the individual portforwarding. Something like a Management Server (Azure etc) with the pi's communicating on port 443. How would this be done? Any advice would be greatly appreciated.
Thanks