Screenly/Anthias #286: Fail to black screen

Cyberprog, 2015-05-18:
Right, we've been having a few issues with Screenly failing to a blank (black) screen overnight. I've enabled all the debug logged (as per my previous request) and have been able to connect to the pi via SSH OK to pull the files off.
So, screenly_view.log starts off 15th of May at 09:18 and runs through our assets until the 17th at 02:51. However there doesn't seem to be anything after this date.
Sun, 17 May 2015 02:50:53 Received non 200-status doesn't appear after the last entry oddly, but does after every other entry.
So I was looking in /tmp and there's a file called screenly_xloader.log, which is *massive* - 252Mb.
This seems to just contain;

Cleaning up...
Traceback (most recent call last):
  File "/usr/lib/python2.7/site.py", line 68, in <module>
    import os
  File "/usr/lib/python2.7/os.py", line 49, in <module>
    import posixpath as path
  File "/usr/lib/python2.7/posixpath.py", line 17, in <module>
    import warnings
  File "/usr/lib/python2.7/warnings.py", line 8, in <module>
    import types
  File "/usr/lib/python2.7/types.py", line 16, in <module>
    IntType = int
NameError: name 'int' is not defined

Repeated over and over again. Last access time on that file is 10:14 on the 18th. (nowish)

I've done a df -h, output below and it does look like we have enough disk space.

pi@BS-Screenly1 /tmp $ df -h
Filesystem      Size  Used Avail Use% Mounted on
rootfs          7.3G  2.5G  4.5G  36% /
/dev/root       7.3G  2.5G  4.5G  36% /
devtmpfs        215M     0  215M   0% /dev
tmpfs            44M  236K   44M   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            88M     0   88M   0% /run/shm
/dev/mmcblk0p1   56M  9.5M   47M  17% /boot

Last entry in supervisord.log is at system boot;

2015-05-15 09:17:20,608 CRIT uncaptured python exception, closing channel <POutputDispatcher at 20005832 for <Subprocess at 19591536 with name screenly in state RUNNING> (stdout)> <snipped as it breaks formatting>
2015-05-15 09:17:20,879 INFO exited: screenly (exit status 245; not expected)
2015-05-15 09:18:01,916 CRIT Supervisor running as root (no user in config file)
2015-05-15 09:18:01,924 WARN Included extra file "/etc/supervisor/conf.d/screenly.conf" during parsing
2015-05-15 09:18:03,342 INFO RPC interface 'supervisor' initialized
2015-05-15 09:18:03,343 WARN cElementTree not installed, using slower XML parser for XML-RPC
2015-05-15 09:18:03,347 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2015-05-15 09:18:03,366 INFO daemonizing the supervisord process
2015-05-15 09:18:03,399 INFO supervisord started with pid 2197
2015-05-15 09:18:04,453 INFO spawned: 'screenly' with pid 2201
2015-05-15 09:18:05,476 INFO success: screenly entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

The whole system seems to stay up and running for about 2 days, then locks up.

Any help/advice would be appreciated! I've not rebooted this system yet if anyone has any ideas to try on it...

vpetersson, 2015-05-18:
Hmm, this device cannot be healthy. `screenly_xloader.log` shouldn't really log that much, so it really shouldn't really grow that fast.

That said, what you posted from the content points to that something is severely broken. As far as I can tell, the errors are not coming from Screenly, but rather from Python itself. Hence, my best guess would be that the system itself is broken.

Cyberprog, 2015-05-18:
So chuck a fresh image at the SD?

vpetersson, 2015-05-18:
Yes, ideally on a brand new SD card as it might be hardware failure.

Before you do that, can you post the content of `/var/log/error`?

Cyberprog, 2015-05-18:
No such file Victor?

vpetersson, 2015-05-18:
Ok, just try flashing it out again then.

Cyberprog, 2015-05-18:
Right-o, will do.

Cyberprog, 2015-05-19:
Do you want an image of this SD card to investigate further Viktor?

vpetersson, 2015-05-19:
Thanks, but I'm OK.

Cyberprog, 2015-05-19:
No worries. I've re-imaged to a fresh card, I wonder if the large image issue has caused these SD's to be worn out prematurely.

vpetersson, 2015-05-19:
It's a reasonable hypothesis. SD cards are pretty fragile, but the quality varies a lot between brands and models. In general, they are more sensitive to writes however.

Cyberprog, 2015-05-19:
Though if we're thrashing /tmp/ with data it might be problematic. Unless /tmp/ is a ramdisk?

vpetersson, 2015-05-19:
Yes, `/tmp` should be ramdisk,

vpetersson, 2015-10-05:
@Cyberprog we can't reproduce this. 

If you have any steps to reproduce them, please post them here and we'll re-open the issue.

closed 2015-10-05 by vpetersson

urb5tar, 2019-05-08:
This issue is still existent. it works in a previous version. But in the current version its broken.

Steps to reproduce

edit /etc/fstab and add line 

`none        /var/log        tmpfs   size=5M,noatime         00`

reboot 

after that the webpage to manage the assets doesn't work.