Screenly/Anthias #287: French translation

forum6691, 2015-05-24:
Is it possible to add the capacity to change the language of the human interface. This could be add in the configuration window for example.

I can translate in french the text of the web pages and buttons if you want.

vpetersson, 2015-05-25:
Localization is something that we haven't really started thinking of yet. It's likely something that we'll do down the road, but it's somewhat complicated and adds a lot of overhead going forward. 

First we'd have to figure what the best localization approach is, and then first implement that in English. As such, it is a fairly large task to undertake.

vpetersson, 2015-05-25:
If you want, you can obviously just fork the project and manually translate the strings there, but that's not something that we'd be able to merge in unfortunately.

closed 2015-07-07 by vpetersson

forum6691, 2022-12-13:
Hi Victor.
Seven years ago (a long time now), I asked for you a French version of screenly OSE.
Today , I propose me again to make a translation in French of the Anthias version (new name) if ther is a xml file with the different char strings used in the software .
Do you think it was possible ?