Screenly/Anthias #289: Duration Setting Being Ignored

skipperut, 2015-05-26:
We are setting a demo of the screenly-ose and have things working well but with a minor issue.  I'm having an issue with the assets not rotating when they are supposed to based off of the duration setting.  I've got 3 assets.  2 images (98KB each) and then the Big Buck Bunny video file that came with the install.  I've set the 2 image files to have a duration of only 5 seconds.  The video file plays for the correct duration before going to the next asset.  When the video file finishes, there's a black screen for 30 seconds before loading the next asset, which I'm not sure why this is being presented .  However, for the 2 assets that are just image files, instead of displaying for 5 seconds, they are being displayed for 45 seconds each. I've ran the ugprade.sh script and rebooted but still experiencing the same result.  Below is the output from the screenly_viewer.log file.  Thanks in advance for your help!

ue, 26 May 2015 15:40:26 Unable to retreive latest SHA
Tue, 26 May 2015 15:40:26 get_next_asset
Tue, 26 May 2015 15:40:26 refresh_playlist
Tue, 26 May 2015 15:40:26 refresh: counter: (0) deadline (2015-06-02 15:33:00) timecur (2015-05-26 19:40:26.355673)
Tue, 26 May 2015 15:40:26 get_next_asset after refresh
Tue, 26 May 2015 15:40:26 get_next_asset counter 0 returning asset 1 of 3
Tue, 26 May 2015 15:40:26 Showing asset Big Buck Bunny Trailer (video)
Tue, 26 May 2015 15:40:26 Asset URI /home/pi/screenly_assets/13dd4bacbb354fa89ebfeb223f9ea2b7
Tue, 26 May 2015 15:40:26 Displaying video /home/pi/screenly_assets/13dd4bacbb354fa89ebfeb223f9ea2b7 for 33
Tue, 26 May 2015 15:40:26 current url is file:///tmp/screenly_html/black_page.html
Tue, 26 May 2015 15:41:00 Last update: 2014-04-25 14:47:05.915333
Tue, 26 May 2015 15:41:40 Unable to retreive latest SHA
Tue, 26 May 2015 15:41:40 get_next_asset
Tue, 26 May 2015 15:41:40 refresh_playlist
Tue, 26 May 2015 15:41:40 refresh: counter: (0) deadline (2015-06-02 15:33:00) timecur (2015-05-26 19:41:40.717121)
Tue, 26 May 2015 15:41:40 get_next_asset after refresh
Tue, 26 May 2015 15:41:40 get_next_asset counter 0 returning asset 2 of 3
Tue, 26 May 2015 15:41:40 Showing asset hok (image)
Tue, 26 May 2015 15:41:40 Asset URI /home/pi/screenly_assets/15d1f8178f7b4c33b98cc74945f382cd
Tue, 26 May 2015 15:41:40 Already showing file:///tmp/screenly_html/black_page.html, keeping it.
Tue, 26 May 2015 15:41:40 Sleeping for 5
Tue, 26 May 2015 15:41:45 Last update: 2014-04-25 14:47:05.915333

vpetersson, 2015-05-26:
Sounds like a bug that we fixed a while back. Try running the update and that should hopefully fix it.

skipperut, 2015-05-26:
That was it.  There was an error when I ran the update (network related) command that I did not catch.  Once I corrected that, the issue was resolved after updating the software and OS.  Thanks for your help!

vpetersson, 2015-05-26:
Glad you got it working.

closed 2015-05-26 by vpetersson

forum6691, 2016-10-17:
Hi everybody
I just install OSE and update all (OSE and system) and I can't fix duration for video. The delay is not memorized and all the video plays the lenght of the video. Impossible for example to play only the fisrt 10 seconds!

Is it a knew bug ?

vpetersson, 2016-10-18:
@forum6691 This by design, not a bug.

forum6691, 2016-10-18:
Thank Victor for your quick answer, but I don't understand it.
It is written on the manage page of screenly that if you "use 0 to autodetect video length", this counts for the duration of all the video and if you enter an other value for the duration, the video plays just the duration entered.

When I read you answer, I understand que the only ablity is to play the whole duration of the video whatever the value entered (none, 0 or any value).  
Could explain me more please ?

vpetersson, 2016-10-19:
Yes, you can only play the full video duration. You cannot only play a part of a video.

forum6691, 2016-10-19:
Understood