Screenly/Anthias #291: Memory leak in uzbl-core?

andersekbom, 2015-06-02:
Hi!

The short of it:
It seems like uzbl-core leaks memory, after restarting it eats up +90% memory in ~15 minutes, which makes the OS start swapping like crazy, which in turns thrashes the SD card after just a few days.

The long of it:
After upgrading to the 2015-02-25-release our Screenly has started to die (display goes black) after 12-18 hours of running. Sometimes restarting the Pi has temporarily made assets show up again, but after 2-3 restarts the Pi won't boot at all, complaining that it's "unable to mount root fs."

Re-flashing on a new high-quality (at least expensive) 8GB SD card repeats the process: it runs for 1-2 days, then starts going black, then crashes on boot.

Running top via SSH (it took a looong time to log in) says that kswapd0 is taking ~100% CPU. uzbl-core takes roughly 90% memory, which I guess is what is causing the swapping.

After killing uzbl-core it restarts and begins eating up memory, after ~15 minutes it's at 90% memory, and the swapping begins anew.

Any log files I could post here to pinpoint the problems?

vpetersson, 2015-06-02:
@andersekbom What kind of web assets are you displaying? If it's a heavy web page, I'm not all that surprised by the behavior unfortunately.

andersekbom, 2015-06-02:
Three web pages, none of them very JS-heavy. Will change to other pages to see if it makes a difference.

ci5ic, 2015-07-13:
I'm getting the same issue on several installations, but I'm not displaying any web assets on these... they are local image assets only.

akanadocs, 2015-09-10:
I'm seeing similar behavior, displaying two very light pages, with minimal JS.  uzbl-core is consuming all my CPU and 90+% memory.  Any solution?

akanadocs, 2015-09-12:
I fix this by using the manual installation instructions to take the latest raspberrian and screenly-ose and my Screenly is now stable without uzbl-core growing.  It would only run for a couple of hours before it died.  Seems rock solid now.

vpetersson, 2015-10-12:
@akanadocs So you're saying in #320 that even installing `uzbl-core` from scratch won't solve the issue. Is this correctly interpreted?

akanadocs, 2015-10-12:
@vpetersson - that is correct yes.  Sorry, I should have come back and updated my earlier post.  It seems to have something to do with the type of web assets being displayed.  It was stable, but then started growing again.

vpetersson, 2015-10-12:
I see. Ok, I will close out this issue for now then since we don't have any obvious solution (here since it's an issue upstream). 

It does only seem to affect some users with certain types of web assets. The workaround for them I would recommend is something hacking like putting `killall uzbl` every night in `crontab` or similar.

Aside from patching `uzbl` upstream (which we don't have resources for), the only *real* solution that I can think of is #257.

closed 2015-10-12 by vpetersson