Screenly/Anthias #292: after sudo apt-get dist-update  blank screen

lenny76, 2015-06-04:
Hi
I've done a system update running apt-get dist-update and apt-get update 

now after the reboot the only thing on screen is a blank screen (after the normal text boot things)
this happens on Rpi1 and Rpi2

any suggestion on how to resolve?

thanks

vpetersson, 2015-06-05:
Something in the setup probably gets overwritten when running `apt-get dist-upgrade`. I'd say you should avoid running that. `apt-get update && apt-get upgrade` should be fine, but even that could cause issues. 

It's really hard to pinpoint what it can be, but perhaps if you re-run the installer, that could solve the issue.

lenny76, 2015-06-05:
tried to reinstall screenly running ./install.sh but without success :(

vpetersson, 2015-06-05:
Ok. I unfortunately do not have access to my lab right now. Hence, the only recommendation I have is to download the disk image and simply not run dist-upgrade. :)

closed 2015-06-05 by vpetersson