Screenly/Anthias #293: Black Screen after Video Asset

skipperut, 2015-06-11:
There's about a 4-6 second delay from when the video asset ends and the next asset begins due to the ile:///tmp/screenly_html/black_page.html setting.  Is there any way to bypass that or shorten that window between the video asset and the next asset?

vpetersson, 2015-06-11:
4-6s isn't all that rare unfortunately if you're dealing with larger videos. The things to try are as follows: 

 * Switch to a faster SD card
 * Use a Raspberry Pi 2

closed 2015-06-11 by vpetersson

skipperut, 2015-06-11:
Interesting.  The video file is only 20-30MB and we currently are using a rapsberry pi 2 and with a Class 10 UHS-I : 10MB/s minimum data transfer rate microSD card.

vpetersson, 2015-06-14:
I see. Yeah, unfortunately there isn't much we can do as that is the load time for the video. The only way around that is to do pre-buffering, but that's something we don't support atm.