Screenly/Anthias #295: No display - File "/home/pi/screenly/viewer.py", line 369, in <module>

benurbano, 2015-06-25:
Hello All,

I have rebuilt this system several times in the past week and it keeps crashing out.
Raspberry Pi 512mb w/ Screenly 2015-02-25 Image.
apt-get update
apt-get upgrade
sudo ~/screenly/misc/run_upgrade.sh
The only additional software I install is AVAHI for ease of discovery on big networks.

Video asset uploaded and worked, Upon reboot it ceases to function.
Screenly web interface on 8080 shows asset in active menu.
But system info only has HDMI info and system performance menu, nothing printing below that to state current function of program. (asset playing.... etc)

Below is the dump from screenly_xloader

Traceback (most recent call last):
  File "/home/pi/screenly/viewer.py", line 369, in <module>
    main()
  File "/home/pi/screenly/viewer.py", line 355, in main
    wait_for_splash_page(url)
  File "/home/pi/screenly/viewer.py", line 345, in wait_for_splash_page
    logging.debug('Waiting for splash-page. Retry %d') % retries
TypeError: unsupported operand type(s) for %: 'NoneType' and 'int'
Cleaning up...

This just keeps repeating every 5-10 seconds.

Could you please advise if you know any issue's? I'm about to try writing just the screenly image/run screenly update, But this time I'll hold off system update/upgrade.

benurbano, 2015-06-26:
Here is debug output after another attempt...

Rewrite screenly ose to SD
Expand SD/Memory split/Password change/Timezone
Set ntp.conf to ntp server for time (internal network)
Put proxy in environment
Put proxy in apt/apt.conf.d/50raspi
***reboot***
apt-get update && install avahi-daemon

Upon reboot...

[image]

benurbano, 2015-07-01:
Problem identified - 
After doing a few rebuilds and trying somethings out. I seemed to have solved my problem.

In /etc/environment - 
I place export http_proxy for internal network.
Upon removing this and doing a reboot, Screenly comes back fine with playlist behaving like normal.

Obviously, If I re-instate proxy string in environment and do reboot, Nothing will display and errors will come out of /tmp/screenly_xloader.

Currently running no proxy in enviroment, But now unable to get screenly webpage.

closed 2015-08-05 by benurbano

vpetersson, 2015-08-05:
Yeah, so proxy isn't supported right now. I'm not sure what the effort would be to add this support.