Screenly/Anthias #296: Multiple connection attempts

GPFisher, 2015-07-06:
We are using Screenly for internal web portals to our organization, and have noticed that in the newest version Screenly is routinely calling pages 5-9 times per minute (back to back), rather than the once per ever 1000 seconds it was configured to. I do know that our page load time is somewhat long at around 30s, but previous versions of Screenly did not exhibit this behavior.

over64, 2015-10-01:
I made some test. I have http server on localhost.
I created one webpage asset with duration 5 sec.
And it works as expected

Serving HTTP on 0.0.0.0 port 3333 ...
10.13.37.117 - - [01/Oct/2015 15:48:24] "HEAD / HTTP/1.1" 200 -
127.0.0.1 - - [01/Oct/2015 15:48:26] "GET /index.html HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:48:59] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:49:34] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:49:40] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:49:45] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:49:50] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:49:55] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:50:00] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:50:05] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:50:10] "HEAD / HTTP/1.1" 200 -

Then I changed duration to 20 sec. Also as expected

10.13.37.117 - - [01/Oct/2015 15:51:11] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:51:31] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:51:51] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:52:11] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:52:31] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:52:51] "HEAD / HTTP/1.1" 200 -
10.13.37.117 - - [01/Oct/2015 15:53:12] "HEAD / HTTP/1.1" 200 -

After this I patched http server for make some network lags.
With 30 sec lag it does not works.

viwer log:
Asset http://10.13.37.120:3333 at http://10.13.37.120:3333 is not available, skipping.

with 5 sec network lag it works as expected. Request period = lag + duration

vpetersson, 2015-10-01:
@GPFisher Closing out this ticket. Please re-open the issue if you're able to provide steps on how to reproduce the issue.

closed 2015-10-01 by vpetersson