Screenly/Anthias #297: Crashing since last update

danbsfc, 2015-07-13:
Both of my Pi's are crashing after 2/3 hours since the last round of updates. Only a reboot gets it back.

See log below...

 Mon, 13 Jul 2015 11:17:36 Loading browser...
Mon, 13 Jul 2015 11:17:36 Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 2330.
Mon, 13 Jul 2015 11:25:06 Generating asset-list...
Mon, 13 Jul 2015 11:25:10 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:25:12 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:27:52 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:27:53 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:30:32 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:30:33 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:33:26 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:33:27 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:36:08 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:36:09 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:38:56 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:38:58 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:41:45 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:41:47 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:44:34 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:44:35 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:47:18 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:47:19 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:50:07 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:50:08 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:52:50 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:52:52 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:55:39 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:55:41 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 11:58:20 Showing asset General Presentation (video)
Mon, 13 Jul 2015 11:58:21 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:01:00 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:01:02 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:03:42 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:03:43 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:06:21 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:06:22 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:09:06 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:09:08 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:11:53 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:11:54 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:14:35 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:14:36 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:17:16 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:17:17 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:19:57 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:19:58 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:22:43 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:22:45 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:25:26 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:25:28 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:28:12 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:28:13 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:30:54 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:30:56 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:33:40 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:33:42 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 12:36:34 Showing asset General Presentation (video)
Mon, 13 Jul 2015 12:36:36 current url is file:///tmp/screenly_html/black_page.html
Mon, 13 Jul 2015 14:17:41 Loading browser...
Mon, 13 Jul 2015 14:17:41 Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 3136.
Mon, 13 Jul 2015 14:17:46 Scheduler init
Mon, 13 Jul 2015 14:17:46 update_playlist
Mon, 13 Jul 2015 14:17:46 Generating asset-list...
Mon, 13 Jul 2015 14:17:46 generate_asset_list deadline: 2030-01-13 15:44:00
Mon, 13 Jul 2015 14:17:46 update_playlist done, count 1, counter 0, index 0, deadline 2030-01-13 15:44:00
Mon, 13 Jul 2015 14:17:46 Entering infinite loop.
Mon, 13 Jul 2015 14:17:46 Last update: 2015-06-19 10:57:45.503958
Mon, 13 Jul 2015 14:17:48 Received non 200-status
Mon, 13 Jul 2015 14:17:48 get_next_asset
Mon, 13 Jul 2015 14:17:48 refresh_playlist
Mon, 13 Jul 2015 14:17:48 refresh: counter: (0) deadline (2030-01-13 15:44:00) timecur (2015-07-13 14:17:48.328315)
Mon, 13 Jul 2015 14:17:48 get_next_asset after refresh
Mon, 13 Jul 2015 14:17:48 get_next_asset counter 0 returning asset 1 of 1
Mon, 13 Jul 2015 14:17:48 Showing asset General Presentation (video)
Mon, 13 Jul 2015 14:17:48 Asset URI /home/pi/screenly_assets/2032e88fb03d4cffa4b4b906d30b7b74
Mon, 13 Jul 2015 14:17:48 Displaying video /home/pi/screenly_assets/2032e88fb03d4cffa4b4b906d30b7b74 for 154 
Mon, 13 Jul 2015 14:17:50 current url is file:///tmp/screenly_html/black_page.html

ci5ic, 2015-07-13:
Getting same. Something is killing my pi's and thrashing the SD cards!!!

danbsfc, 2015-07-14:
Gone back to an old backup this morning and all is good. Been up and running since 8:30 without any issues. Must be something in the update.

akanadocs, 2015-09-12:
I took the opposite approach and used the manual installation instructions to take the latest raspberrian and screenly-ose and my Screenly is now stable without uzbl-core growing.  It would only run for a couple of hours before it died.  Seems rock solid now.

vpetersson, 2015-09-14:
@akanadocs Interesting find. I suppose we should make an updated disk image that way (with some minor modifications that we do to the disk images).

vpetersson, 2015-10-12:
Closed in favor of #320.

closed 2015-10-12 by vpetersson