Screenly/Anthias #298: Support for looping video & orientation

abrenoch, 2015-07-21:
We have recently found a need to play looping video (in portrait mode) with screenly.

I know this isn't supported natively, but the most recent versions of omxplayer do actually support changing the video orientation & a loop flag.

In the meantime I am just turning off all the assets, and playing the video with the orientation & loop flags via SSH (with an updated version of omxplayer). I have not seen any impact on performance.

vpetersson, 2015-07-21:
Interesting observation. Thanks, @abrenoch. We'll take a look.

abrenoch, 2018-02-09:
Just wanted to bump this - unless I'm mistaken I believe SSH is still the only way to accomplish changing orientation & seamless looping!

vpetersson, 2018-02-09:
SSH is the only way to sort out orientation, that is correct.

As far as the loop feature goes, that would require development work. I haven't looked into this, but I presume it would require something like conditional checks in the playlist and then appending some flag if there is only one video asset (which likely is an edge case IMHO).

joeladria, 2019-02-05:
@abrenoch what exactly did you modify in SSH to achieve this?

abrenoch, 2019-02-05:
> @abrenoch what exactly did you modify in SSH to achieve this?

Originally I had to build and install a newer version of omxplayer, but it appears that the more recent versions of screenly do support the `--loop` and `--orientation` flags.

Basically I would just call out the screenly asset I want to play looped & rotated, launching it with omxplayer. Provide the orientation & loop flags as needed to make it display/play as expected. This needs to be done every time you want to play a video.

Kinda stinks but will get the job done if you are in a pinch!

joeladria, 2019-05-13:
I've created a fork that rather crudely implements these features as global functions. Unfortunately it breaks scheduling, but if you're just looking for a quick version that does looping and orientation this will do the trick. https://github.com/joeladria/screenly-ose