Screenly/Anthias #302: Screenly git remote using git:// makes updating behind proxy difficult

wlcx, 2015-08-05:
If the remote used the http transport, git would just use the `HTTP_PROXY` envar. Is there any reason it's `git://`?

vpetersson, 2015-08-05:
Well, pulling over `ssh` is just more common. Besides, there is no full proxy support in Screenly at the moment, so perhaps it's better that it breaks at that stage than random behavior later.

wlcx, 2015-08-05:
`git://` isn't ssh though, [it's a separate protocol](https://git-scm.com/book/en/v2/Git-on-the-Server-The-Protocols#The-Git-Protocol) - I know there isn't out of the box proxy support but everything else is configured pretty easily with `HTTP(S)_PROXY` in /etc/environment - I just wondered if there was a particular reason. Also it would be one step closer to full proxy support :stuck_out_tongue:

vpetersson, 2015-08-05:
Ah, good point regarding the Git protocol. My bad. I mixed it up with the SSH encapsulated version.

I will however close out this ticket in favor of #158, as it doesn't make a ton of sense to add this before then.

closed 2015-08-05 by vpetersson