Screenly/Anthias #304: Wifi support

romspax, 2015-09-21:
Hi,

I think this should be labeld as a question/feature request. In the FAQ i read that there's no wifi support and i'm baffeld by this. Is there REALLY no option to use ''normal'' wifi (other than a wifi bridge/extender-solution described in the FAQ?

ViktorNova, 2015-09-23:
You absolutely can use WiFi.
It is Debian Linux (Raspbian) after all under the hood, which works as well with WiFi as anything else, I'm not sure why their official word says otherwise.

    sudo apt-get install network-manager

And then use [nmcli](http://www.thepromisedlan.org/archives/nmcli-is-wicked) to connect.

When a newer version of Screenly based on the current version of Debian (Jessie) comes out we can easily use ````nmtui```` to configure networks (including WiFi) but for now you are stuck with nmcli or the old fashioned way using wpa-supplicant

You can also use [wicd-curses](http://www.raspyfi.com/wi-fi-on-raspberry-pi-a-simple-guide) as an alternative terminal GUI for configuring WiFi
(do ctrl+alt+F1 to switch to a tty console and log in)

vpetersson, 2015-09-23:
Screenly OSE does support WiFi, since you're basically installing it on a regular Raspbian setup. Screenly Pro however, does not support WiFi.

There's no need for the tools suggested by @ViktorNova. All you need to do is to add the following stanza to `/etc/network/interfaces`:

```
allow-hotplug wlan0
auto wlan0
iface wlan0 inet dhcp
        wpa-ssid "ssid"
        wpa-psk "password"
```

closed 2015-09-23 by vpetersson

ViktorNova, 2015-10-12:
@vpetersson 
Wow, for all my years of running linux I somehow completely failed to realize it was actually this simple to do WiFi in the terminal.. thanks!

vpetersson, 2015-10-12:
Yeah, there are lots of strange suggestions on the Raspberry Pi forum, but this is by far the easiest solution IMHO.

johnnyrevell, 2017-03-06:
Tested for WPA2 / Linux raspberrypi 4.4.34+ #930 Wed Nov 23 15:12:30 GMT 2016 armv6l GNU/Linux
1. su root
`sudo su`
2. Install nmcli 
`apt-get install network-manager`
3. Optional - scan network
`nmcli dev wifi list`
4. Add wireless network
`nmcli con add con-name NAME_FOR_CONNECTION ifname wlan0 type wifi ssid WIFI_SSID`
5. Tell nm connection is WPA
`nmcli con modify NAME_FOR_CONNECTION  wifi-sec.key-mgmt wpa-psk`
6. Store wifi password
`nmcli con modify bcprep  wifi-sec.psk WIFI_PASSWORD`
7. Bring up connection
`nmcli con up bcprep`

Should survive reboot- check with `nmcli connection show --active`

vpetersson, 2017-03-06:
@johnnyrevell That is a valid approach, but i would still recommend using `/etc/network/interfaces` instead as it doesn't require any new dependencies.

rwwest7, 2017-04-29:
@vpeterson When I try to use /etc/network/interfaces, every time I reboot Screenly OSE erases any changes I had made at reboot.

The network-manager method worked great though.

vpetersson, 2017-04-29:
> When I try to use /etc/network/interfaces, every time I reboot Screenly OSE erases any changes I had made at reboot.

Yup. This is a side-effect as network manager does overwrite it as part of the setup. We have also created a [wizard](https://wizard.screenly.io/) to simplify the process of creating this file.

rwwest7, 2017-05-07:
I loaded up a new Pi, and used the wizard to create the network.ini file. Worked great.

But, after I updated Screenly OSE the file was overwritten with the default one again. After thinking my wireless was dropping I finally looked at the file and all my changes were lost.

Is there a way to run the upgrade without losing all settings?

vpetersson, 2017-05-07:
Yeah, you need two switch to the development version for now.

boschie, 2017-05-10:
If the settings were written in the /boot/network.ini file, are they also overwritten after an upgrade?

rwwest7, 2017-05-11:
> If the settings were written in the /boot/network.ini file, are they also overwritten after an upgrade?

@boschie 
Yes. I downloaded the latest image available on the releases page. Used the wizard to create the network.ini file. Put it in the boot folder. Worked great.

The web gui told me an update was available, so I ran the upgrade command ($ bash <(curl -sL https://www.screenly.io/install-ose.sh)). After a lengthy upgrade, my /boot/network.ini file was replaced with the default one.

Also, if I put my original network.ini back it seems to ignore it and no wifi. If I use johnnyrevell's nmcli method from above then wifi works great.

epers, 2018-01-24:
Noticed that screenly-net-manager.service was overwriting /etc/network/interfaces with the settings set in /boot/network.ini

If you're like me and would rather write your own interfaces file, all you have to do to keep screenly from overwriting it on you each boot is

`systemctl disable screenly-net-manager.service`

vpetersson, 2018-01-24:
@epers FYI - We've decommission Screenly Network Manager in the latest release.

ACEDornback, 2019-12-18:
So I went through the 7 steps above.
I now see the connection with an nmcli connection show but device is listed as --
And I received an error on step 7 when I try to bring up the connection.

"Error: Connection activation failed: No suitable device found for this connection."

It is a USB wifi device that I used with previous Screenly versions as well as other Raspian and other OS on this same device.

It is as though the drivers/framework is not fully/properly loaded to support a USB wifi.
However, I can view the device AP/SSID when I boot it up but cannot connect to the wifi.

vpetersson, 2019-12-19:
> It is a USB wifi device that I used with previous Screenly versions as well as other Raspian and other OS on this same device.

Hard to say. We don't really do anything with regards to network drivers. If I'm not mistaken, some WiFi drivers (either due to lack in the driver or hardware) are unable to act as an access point. Perhaps that's it.

The easiest workaround is probably to manually configure the WiFi to get around this.

ACEDornback, 2019-12-19:
I flashed with a previous version and it is working over wifi with a wifi USB.