Screenly/Anthias #306: Resolve Auth pull request

vpetersson, 2015-09-23:
We should finalize this pull request:
https://github.com/wireload/screenly-ose/pull/224

mrneilsy, 2015-09-23:
+1. For an organisation of my size, with the potential to deploy dozens of Screenly OSEs, secure LDAP access control is a clincher. We're using the server.py auth over SSL for now. Is the LDAP ready to test? If not is it something you could give a tentative ETA for?

vpetersson, 2015-09-23:
LDAP is part of the PR, but the top priority would be to get regular auth working and part of the system.

vpetersson, 2016-05-03:
I'm torn if we should go down the route of having this natively or doing it in Nginx (see #402). 

The benefit of doing it in Nginx is that it is a lot simpler (just basic auth). 

The downside that it doesn't support things like LDAP.

mrneilsy, 2016-12-02:
Is this still in pipeline? Basic / LDAP?

vpetersson, 2016-12-02:
@mrneilsy we're going down the route of #495 right now. it's a less complex change that wouldn't require a huge refactoring.

LDAP is a bit more complex (by the nature of the protocol).

vpetersson, 2017-07-01:
Closing this one out as this has been sorted in a different way.

closed 2017-07-01 by vpetersson