Screenly/Anthias #320: Create new disk image

vpetersson, 2015-10-12:
In light of the discussion in #297, we should create a new disk image based on the latest Raspbian.

akanadocs, 2015-10-12:
Just a note here.  I'm not sure that my solution - latest Raspbian plus latest screenly-ose has actually worked.  I'm still getting massive memory growth.  What I've done now is to use a cron job to restart the device at 5am every day.  The growth is all in the uzbl-core process.

vpetersson, 2015-10-12:
Hmm. Ok. Can you provide a dump such that we can take a closer look and try to reproduce it?

```
tar cvfz dump.tgz /home/pi/.screenly /home/pi/screenly
```

akanadocs, 2015-10-12:
I can't upload the file - Github won't accept zipped tar archives.  It's only 1.5Mb, I'm just not sure of the best way for me to get it to you.

vpetersson, 2015-10-12:
Can you post it on Dropbox (or something similar) and put the link here?

akanadocs, 2015-10-12:
You can grab it here: https://github.com/akanadocs/assets/blob/master/dump.tgz

Let me know if you need anything else.  Maybe a core dump of the uzbl-core process.

vpetersson, 2015-10-12:
We'll take a look and see if there's anything obvious that we can look into. Unfortunately my guess is that it isn't (as pointed out in #291).

vpetersson, 2016-04-26:
New image si now available.

closed 2016-04-26 by vpetersson