Screenly/Anthias #321: Implement integration tests in Travis

vpetersson, 2015-10-12:
There are some old integration tests in the system already. However, they are not being tested by Travis yet. The objective of this issue is to implement this such that these tests are being executed (and pass) on Travis.

vpetersson, 2016-07-12:
We should use PhantomJS for this. There is documentation for this [here](https://docs.travis-ci.com/user/gui-and-headless-browsers/).

over64, 2016-08-19:
Achieved this on localhost with Phantomjs:
```
over@over-debby14:~/build/gf/screenly-ose-gf/static/spec$ phantomjs prunner.js
Runner Started.
Screenly Open Source : should have a Screenly object at its root ... 
Passed.
date_to : should format date and time as 'MM/DD/YYYY hh:mm:ss A' ... 
Passed.
date_to : should format date as 'MM/a_date/YYYY' ... 
Passed.
date_to : should format date as 'hh:mm:ss A' ... 
Passed.
date_to: 3 of 3 passed.
Asset model : should exist ... 
Passed.
Asset model : should be active if enabled and date is in range ... 
Passed.
Asset model : should be inactive if disabled and date is in range ... 
Passed.
Asset model : should be inactive if enabled and date is out of range ... 
Passed.
Asset model : should rollback to backup data if it exists ... 
Passed.
Asset model : should erase backup date after rollback ... 
Passed.
Asset model: 12 of 12 passed.
Models: 12 of 12 passed.
Assets : should exist ... 
Passed.
Assets : should use the Asset model ... 
Passed.
Assets : should keep play order of assets ... 
Passed.
Assets: 8 of 8 passed.
Collections: 8 of 8 passed.
Views : should have EditAssetView ... 
Passed.
Views : should have AssetRowView ... 
Passed.
Views : should have AssetsView ... 
Passed.
Views: 3 of 3 passed.
Screenly Open Source: 27 of 27 passed.
Runner Finished.
16 specs, 0 failures in 0.024s.
status: success
```

phantom-runner.js
```javascript
var webPage = require('webpage');
var page = webPage.create();

page.onConsoleMessage = function(msg, lineNum, sourceId) {
  console.log(msg);
};

page.open('http://localhost:8080/static/spec/runner.html', function (status) {
  var url = page.url;
  console.log('status: ' + status);
  phantom.exit(0);
});
```
easy
correct exit codes in process

vpetersson, 2016-08-19:
Cool. Now we need to implement this with Travis and then add some good tests for common use cases.

commit 99631475 (2016-08-31): Merge pull request #476 from GreatFruitOmsk/feture/web-tests-ci

Travis CI for web tests. Closes #321
closed 2016-08-31 by vpetersson