Screenly/Anthias #325: Assets not staying in the order you put it in

mlittleds9, 2015-10-29:
Why does the order change when you close the browser... When you change the position of the Assets it does not save it.. when you go back into the page it ignores what order you had it in...

Why is there no save button because at the moment it is not saving the position of the Assets after you change it... Is there a correct way to save the position of the Assets?

Is this a know problem with the OSE version?

vpetersson, 2015-10-29:
This is not a known issue. It could possibly be related to #229. 

Are you sure you're running on the latest version?

mlittleds9, 2015-10-29:
How can you tell what version you are running? I updated them last night and they are all doing the same thing.

vpetersson, 2015-10-29:
It will say 'Update available' in the web interface if you're not using the latest version.

In any case, if you updated last night, that should be the latest version.

@over64 Could you try to reproduce this?

mlittleds9, 2015-10-29:
Yep, i have three devices running screenly OSE, none of them are saying update available. 

They all do the same thing if you move one of the assets to the bottom and then close the page, when you back into the page the Asset has not moved or sometimes the Asset has moved one position. I am bit surprised there is no Save button at the bottom of the page.

vpetersson, 2015-10-29:
The assets should be automatically saved, which means there is no need for a save button.

vpetersson, 2015-10-29:
I am unable to reproduce this. It autosaves for me on the [demo site](http://ose.demo.screenlyapp.com/). 

What browser, browser version and operating system are you using?

mlittleds9, 2015-10-29:
im using IE 11 in Windows 10, but it also happens in IE II Windows 7...

its really strange, i wonder if it works in Chrome..

vpetersson, 2015-10-29:
Yes, could very well be a IE issue. Try Chrome instead.

mlittleds9, 2015-10-29:
Ok I have tried updating the page in Chrome instead of IE and it does work better in Chrome...

It must be IE...

vpetersson, 2015-10-29:
Frankly, full support for IE is not super high up on the priority list, as it has a declining user base and it's a substantial effort.

closed 2015-10-29 by vpetersson