Screenly/Anthias #326: Load local html files

pelicanmedia, 2015-11-04:
Hi would it be possible for Screenly to load local .html files.
ie:

```` file:///home/pi/localfiles.index.html ````

Thanks

vpetersson, 2015-11-04:
In theroy, yes, but it's not something that makes sense to implement as it would be a nightmare to do in a secure fashion.

closed 2015-11-04 by vpetersson

Costly, 2017-05-17:
Can you take another look at this please? This makes perfect sense for some use-cases. I want to built a dashboard for my amazon sales. The data can be fetched from the Amazon MWS API. 

I have two options of get this into a screenly screen but both include a local html file.

1. I can fetch the data with javascript from a local html file.
2. I can build a script which fetches the data and build a new local html file.

I dont want my API access keys to be hosted somewhere in the internet so using the currently "webpage" feature is not an option.

vpetersson, 2017-05-18:
@Costly A far better approach IMHO is to simply run a lightweight web server (such as Nginx) on either a separate Pi or the Pi running Screenly serving this content. You can configure Nginx to simply bind on localhost if you're concerned about other people on the LAN accessing it.

Costly, 2017-05-19:
That's what i did as a workround. But it feels like really unnecessary overhead. Why "transfer" a file via HTTP when you have access to the local file system?

vpetersson, 2017-05-19:
The overhead is rather minor. Besides, serving HTTP content from disk directly (i.e. `file://`) is bound to break some HTTP/JavaScript/CSS functionalities as the content is not designed to be served this way.