Screenly/Anthias #327: Feature Request: Youtube streaming

JBowness, 2015-11-06:
openelec seems to do a good job of this on a Pi 2... would be incredible to see support for youtube videos/playlists/live feeds in screenly.

(Is this a feature available in pro?)

vpetersson, 2015-11-07:
My initial response to this was that it isn't possible, because the browser we use (`uzbl`), doesn't support HTML5 playback (partially due to lack of hardware acceleration). 

However, after thinking about this for a bit today and doing some research, I came up with an alternative approach. Instead of playing this using the browser, we could threat it as a video object directly. 

The problem with this however was that we need to extract the MPEG4 video feed from YouTube. Fortunately, there is a tool called [youtube-dl](https://github.com/rg3/youtube-dl) that can be used for this. 

Getting the video feed from a YouTube clip is as easy as:

    $ youtube-dl -g https://www.youtube.com/watch?v=fcxsfN4jSV8

(Since `youtube-dl` is written in Python we could do it in a more intelligent fashion, by adding this as a submodule and do some kind of function call probably.)

We could then store this URL in the database and simply play this URL directly with `omxplayer` (which I believe supports http/https playback).

nilsbyte, 2016-03-17:
Any news on this? the suggestion sounds promising.

vpetersson, 2016-03-17:
Nope, but contributions are welcome.

robberwick, 2016-04-25:
@vpetersson  Assuming that `youtube-dl` downloading isn't contravening Google's Terms and Conditions, this sounds like the kind of thing that would be best implemented as an asynchronous task using Celery/Huey/Channels and some sort of queue (probably Redis). Are there any plans to add support for such things?

vpetersson, 2016-04-25:
@robberwick I took a closer look and found [this](https://law.stackexchange.com/questions/1648/is-downloading-youtube-videos-legal) thread. tl;dr - it appears to be somewhat gray area. As such, we won't spend resources on this internally. If however someone would like to do the work, we will likely accept the pull request (assuming that it is solid and properly tested).

closed 2016-04-25 by vpetersson