Screenly/Anthias #329: RTMP Streaming for DAUs

RaspiNewb, 2015-11-19:
Hey Guys,

i just set up Screenly on my Raspi. Everything goes as well and i manually changed the Standard EndTime to 10 Years in
the future in the screenly-ose.js.

So i just read Screenly is able to playback rtmp Links via OMXplayer?

But i dont be able to set up one of these.

I just changed the accepted URL Types in the utils.py file so i could save the rtmp link in the backend but there comes
just this error

Server Error: No connection adapters were found for 'rtmp://pssimn24livefs.fplive.net/pssimn24live-live/_definst_/stream1'

where i have to configure the adapter?

RaspiNewb, 2015-11-20:
i just tried to playback a livestream the normal way without any problems

i just exectued this command via ssh on the screenly

omxplayer rtmp://pssimn24livefs.fplive.net/pssimn24live-live/_definst_/stream1

so the player overlays the screenly 

Question is, how can i put this command into the normal circle for playback within the other
pictures, videos and so on

I also tried to add a new span for "Livestream" in the views/index.haml but it doesnt appear

just anyone who could help me?

brosserg, 2015-11-25:
I was wondering if you had any luck with getting your RTMP stream working? I too would like to get this working.

vpetersson, 2015-11-25:
Added this for Sprint 3.

vpetersson, 2016-05-03:
Consolidating. Duplicate of #90.

closed 2016-05-03 by vpetersson