Screenly/Anthias #334: Scheduled Assets not working

rsenio, 2015-12-04:
Twice now I have scheduled an asset to start at a specific time of day (simple jpeg image) and it doesn't start when it is supposed to. It will only begin if I re-visit the web UI after the asset should be running. 

For example:
scheduled a jpeg to begin at 10:55am to go to 3:00pm.
checked the screen at 10:57 and saw that the asset was not playing
logged into the web UI and could see that it should be running, it immediately began playing

rsenio, 2015-12-04:
Here's a cut from the log.

You'll see that I logged into the webUI at 10:57...the log reports generation asset list. However, this asset was added early this morning, and scheduled + turned on

Fri, 04 Dec 2015 10:54:54 Sleeping for 5
Fri, 04 Dec 2015 10:54:59 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:54:59 Sleeping for 5
Fri, 04 Dec 2015 10:55:04 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:55:04 Sleeping for 5
Fri, 04 Dec 2015 10:55:09 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:55:09 Sleeping for 5
Fri, 04 Dec 2015 10:55:14 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:55:14 Sleeping for 5
Fri, 04 Dec 2015 10:55:19 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:55:19 Sleeping for 5
Fri, 04 Dec 2015 10:55:24 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:55:24 Sleeping for 5
Fri, 04 Dec 2015 10:55:29 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:55:30 Sleeping for 5
Fri, 04 Dec 2015 10:55:35 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:55:35 Sleeping for 5
Fri, 04 Dec 2015 10:55:40 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:55:40 Sleeping for 5
Fri, 04 Dec 2015 10:55:45 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:55:45 Sleeping for 5
Fri, 04 Dec 2015 10:55:50 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:55:50 Sleeping for 5
Fri, 04 Dec 2015 10:55:55 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:55:55 Sleeping for 5
Fri, 04 Dec 2015 10:56:00 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:56:00 Sleeping for 5
Fri, 04 Dec 2015 10:56:05 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:56:05 Sleeping for 5
Fri, 04 Dec 2015 10:56:10 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:56:10 Sleeping for 5
Fri, 04 Dec 2015 10:56:15 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:56:15 Sleeping for 5
Fri, 04 Dec 2015 10:56:20 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:56:20 Sleeping for 5
Fri, 04 Dec 2015 10:56:25 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:56:25 Sleeping for 5
Fri, 04 Dec 2015 10:56:30 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:56:30 Sleeping for 5
Fri, 04 Dec 2015 10:56:35 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:56:35 Sleeping for 5
Fri, 04 Dec 2015 10:56:40 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:56:40 Sleeping for 5
Fri, 04 Dec 2015 10:56:45 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:56:45 Sleeping for 5
Fri, 04 Dec 2015 10:56:50 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:56:50 Sleeping for 5
Fri, 04 Dec 2015 10:56:55 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:56:55 Sleeping for 5
Fri, 04 Dec 2015 10:57:00 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:57:00 Sleeping for 5
Fri, 04 Dec 2015 10:57:05 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:57:05 Sleeping for 5
Fri, 04 Dec 2015 10:57:10 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:57:10 Sleeping for 5
Fri, 04 Dec 2015 10:57:15 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:57:15 Sleeping for 5
Fri, 04 Dec 2015 10:57:20 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:57:20 Sleeping for 5
Fri, 04 Dec 2015 10:57:25 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:57:26 Sleeping for 5
Fri, 04 Dec 2015 10:57:31 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:57:31 Sleeping for 5
Fri, 04 Dec 2015 10:57:36 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:57:36 Sleeping for 5
Fri, 04 Dec 2015 10:57:41 Generating asset-list...
Fri, 04 Dec 2015 10:57:41 Showing asset United Overseas Bank (image)
Fri, 04 Dec 2015 10:57:41 Sleeping for 10
Fri, 04 Dec 2015 10:57:51 Showing asset Keysource (image)
Fri, 04 Dec 2015 10:57:51 Sleeping for 5
Fri, 04 Dec 2015 10:57:56 Showing asset Combustex (image)
Fri, 04 Dec 2015 10:57:56 Sleeping for 5
Fri, 04 Dec 2015 10:58:01 Showing asset Welcome (image)
Fri, 04 Dec 2015 10:58:01 Sleeping for 5
Fri, 04 Dec 2015 10:58:06 Showing asset United Overseas Bank (image)
Fri, 04 Dec 2015 10:58:06 Sleeping for 10

rsenio, 2015-12-04:
Perhaps this is back

https://github.com/wireload/screenly-ose/issues/209

vpetersson, 2015-12-04:
@over64 can you try to see if you can reproduce that?

over64, 2015-12-07:
I had some issues with scheduling twice. New assets was not being added .But I cannot reproduce this. I have only one version - bug with sqlite file synchronization. It seems that file mtime is not good sync primitive. But now it's only versions.

over64, 2015-12-08:
It seems that the same is https://github.com/wireload/screenly-ose/issues/147

rsenio, 2015-12-08:
Not sure if it matters, in terms of replicating the issue. I have two assets that have been playing for the whole year, and scheduled to end way in the future. It is these "one offs" that get scheduled to start and play for a day or two along side the long standing assets, but do not work unless I re-visit the webui.

davezdesignz, 2016-03-30:
Check that your end date is in the future and not in the past or else they do not activate.

rsenio, 2016-03-30:
uhh....yeah, that's pretty self explanatory.

davezdesignz, 2016-03-30:
yeh some people might not think to check it though ... don't worry I won't post anymore.

davezdesignz, 2016-03-30:
apparently it wasn't explanatory if a customer contacted me with the issue of an asset that wouldn't become active and that was their issue ... there was no error, it turned to on but didn't jump up to active, was their end date. Due to your ignorant comment I will find them a new solution!

rsenio, 2016-03-30:
Lmao, relax. If you read the initial post, the asset DOES become active after re-visiting the web ui. Anyway, sorry you're having such a bad day and that 6 words can set you off. However, this isn't the place for it

vpetersson, 2016-03-31:
Closing out as duplicate of #147.

closed 2016-03-31 by vpetersson

rsenio, 2016-04-25:
same thing happened today

vpetersson, 2016-04-25:
Could you please tar up the whole content and we can see if we can reproduce it. Just run:

`tar cvfz ~/screenly_dump.tgz ~/screenly ~/.screenly`

and then attach `~/screenly_dump.tgz` to this issue.

rsenio, 2016-04-25:
After I saw the issue, I re-visited the web UI and it kick started the asset into playing. At that point I saw another update available. I have updated, and rebooted. I then got your email, I’ve run your command below and attached the results. Hopefully me running the update doesn’t mess up gathering the info required.