Screenly/Anthias #337: Web Asset connectivity lost

Rotbart, 2015-12-09:
Hello,

iam trieing to use screenly for Web asset.
I set the reload time to 25 Seconds and tried to unplug the network cable to look what happens.
It shows website cant be found (thats good)

If i replug the cable the web asset is not reloading the page
Is the a trick or fix for that?

the only possibility i found was to load a image/video and then the web asset again.
but i wont be able to look the whole time if the web asset is online or not 

br

Rotbart, 2015-12-14:
found out this seems to be a refresh issue like the post bevor 
if you have just one webasset the refresh is not working

qbicdesign, 2016-01-08:
I hit the same issue
workaround would be to add the same web asset twice I guess
Gonna try it

vpetersson, 2016-04-26:
I wasn't able to reproduce this in the latest version.

closed 2016-04-26 by vpetersson