Screenly/Anthias #340: Web interface not available

Joshy86, 2015-12-18:
Hello,

Since a couple of days playing around with a new Raspberry Pi 2B and Screenly.
Today, suddenly I could not connect anymore to the web interface on my local LAN (IP:8080). 
Restored an image to SD card, everything works again and checked for updates "sudo ~/screenly/misc/run_upgrade.sh".
Rebooted and again, no connection with my webbrowser. SSH with Putty still  works and the screen is still on.
Just again, no web interface connection.

Am I doing something wrong? Or is this a bug? 

Thanks for helping.. 
Josh (newbie)

jscaptura, 2015-12-18:
Since installing the latest update this morning, I am having the exact same issue. 

Here's my upgrade log:

pi@screenly ~ $ ~/screenly/misc/run_upgrade.sh
Upgrading Screenly OSE...
Ensuring proper permission is set...
Removing feh (no longer needed)...
Installing libx11-dev (if missing)...
Removing OS-provided supervisor...
Stopping supervisor: supervisord.
(Reading database ... 65376 files and directories currently installed.)
Removing supervisor ...
Stopping supervisor: supervisord.
Purging configuration files for supervisor ...
dpkg: warning: while removing supervisor, directory '/etc/supervisor/conf.d' not empty so not removed
dpkg: warning: while removing supervisor, directory '/var/log/supervisor' not empty so not removed
Processing triggers for python-support ...
Installing supervisor from pip packages
update-rc.d: using dependency based boot sequencing
update-rc.d: error: unable to read /etc/init.d/supervisor
sudo: /etc/init.d/supervisor: command not found
Removing 'unclutter' and replacing it with a better hack.
unclutter: no process found
Fetching the latest update...
remote: Counting objects: 23, done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 23 (delta 8), reused 4 (delta 4), pack-reused 9
Unpacking objects: 100% (23/23), done.
From git://github.com/wireload/screenly-ose
   911380f..ffe427d  master     -> origin/master
Updating 911380f..ffe427d
Fast-forward
 misc/install.sh       |   11 +++-
 misc/supervisor       |  167 +++++++++++++++++++++++++++++++++++++++++++++++++
 misc/supervisord.conf |   28 +++++++++
 misc/upgrade.sh       |   15 +++++
 4 files changed, 219 insertions(+), 2 deletions(-)
 create mode 100755 misc/supervisor
 create mode 100644 misc/supervisord.conf
Ensuring all Python modules are installed...
Running migration...
Obsolete column (filename) is not present
Columns (is_enabled,nocache) already present
already has primary key
Column (play_order) already present
Migration done.
Restarting app-server...
error: <class 'socket.error'>, [Errno 97] Address family not supported by protocol: file: /usr/lib/python2.7/socket.py line: 571
Restarting X (viewer)...
[ ok ] Stopping Light Display Manager: lightdm.
[ ok ] Starting Light Display Manager: lightdm.
Done! Please reboot.
pi@screenly ~ $

Jon (experienced user)

vpetersson, 2015-12-19:
This looks suspicious: 
    error: , [Errno 97] Address family not supported by protocol: file: /usr/lib/python2.7/socket.py line: 571

It is likely related to the supervisor upgrade. Looks like there's an open [Issue](https://github.com/Supervisor/supervisor/issues/376) for it. 

@over64 Can you take a look?

over64, 2015-12-19:
Could you provide supervisor log at /var/log/supervisor/ ?

vpetersson, 2015-12-19:
@over64 Judging by the response in the issue I linked to, I think the solution is to change [:49](https://github.com/wireload/screenly-ose/blob/master/misc/upgrade.sh#L49) in `upgrade.sh` from:

    sudo supervisorctl restart screenly

to

    sudo supervisorctl -c /etc/supervisord.conf restart screenly

If I'm not mistake, we had a similar issue in Screenly Pro, and it was solved by symlinking     `/etc/supervisor/supervisord.conf` to `/etc/supervisord.conf`, which could be a cleaner solution.

In either case, I don't think you need the logs, as the error should be easy to reproduce.

over64, 2015-12-20:
@vpetersson, found problem. Will make patch in 3 hours
@jscaptura as simple workaround you can just run upgrade.sh again. It should fix problem

Frankstar, 2015-12-21:
Just updated vida upgrade.sh -> still the same Problem.
error: <class 'socket.error'>, [Errno 97] Address family not supported by protocol: file: /usr/lib/python2.7/socket.py line: 571
Restarting X (viewer)...

vpetersson, 2015-12-21:
@Frankstar Yeah, fix is on the way (#342).

Frankstar, 2015-12-21:
fastest replie ever ...
Ty @vpetersson  :)

jscaptura, 2015-12-21:
Running upgrade.sh a second time corrected the problem.

-- Jon

Frankstar, 2015-12-21:
@jscaptura interresting.
I get the same error over an over again.
Waiting for the fix :)

vpetersson, 2015-12-21:
@Frankstar try this as a temporary workaround:

    sudo supervisorctl -c /etc/supervisord.conf restart screenly

Frankstar, 2015-12-21:
@vpetersson 
i tryed already. but i got 
> Error: could not find config file /etc/supervisord.conf  

but:
> sudo supervisorctl -c /etc/supervisor/supervisord.conf screenly restart

worked. ty

vpetersson, 2015-12-21:
Ah, right. Good catch.

closed 2015-12-21 by vpetersson

GeoHolz, 2016-01-12:
Hello, 
I have the same problem but run run_upgrade.sh again and again don't solve the problem and

pi@screenly ~/screenly/misc $ sudo supervisorctl -c /etc/supervisor/supervisord.conf restart screenly
Error: could not find config file /etc/supervisor/supervisord.conf
For help, use /usr/local/bin/supervisorctl -h

Can u help me ?

Frankstar, 2016-01-12:
actually this issue is already fixed in Master https://github.com/wireload/screenly-ose/pull/342
I didnt run in this Problem anymore.

vpetersson, 2016-01-12:
@GeoHolz Sounds like something went wrong with either the installation or upgrade. Try running the upgrade.

chvostek, 2016-09-27:
Yet another update to this process.

My up-to-date Screenly players appear to use **systemd** to make sure things start and are running. Thus, rather than killing off the process and hoping it will come back, or trying to use `supervisorctl` which is no longer part of the system, you can force a restart of the viewer with the following command:

    $ sudo systemctl restart screenly-viewer

If you turn off "Show splash screen" in settings, then this command will take 2 to 3 seconds to force a restart of your playlist.

You can look at `/etc/systemd/system/screenly-viewer.service` to see how systemd handles screenly.