Screenly/Anthias #344: Internal server error 500 when i go to System_info

Thrakon, 2015-12-23:
Hi!

I've been using Screenly at work for about 6 days straigt, and now i get an Internal Server Error (500) when attempting to go to the System_Info page.

The error message goes as follows;
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 50: ordinal not in range(128)
100.231.13.87 - - [23/Dec/2015 10:28:02] "GET /system_info HTTP/1.1" 500 756
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1729, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/screenly/server.py", line 319, in system_info
    return template('system_info', viewlog=viewlog, loadavg=loadavg, free_space=free_space, uptime=system_uptime, display_info=display_info)
  File "/home/pi/screenly/server.py", line 97, in template
    return haml_template(template_name, **context)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 3592, in template
    return TEMPLATES[tplid].render(kwargs)
  File "/usr/local/lib/python2.7/dist-packages/bottlehaml/__init__.py", line 34, in render
    return self.tpl.render(**_defaults)
  File "/usr/local/lib/python2.7/dist-packages/mako/template.py", line 412, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/local/lib/python2.7/dist-packages/mako/runtime.py", line 766, in _render
    **_kwargs_for_callable(callable_, data))
  File "/usr/local/lib/python2.7/dist-packages/mako/runtime.py", line 798, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/local/lib/python2.7/dist-packages/mako/runtime.py", line 824, in _exec_template
    callable_(context, *args, **kwargs)
  File "system_info", line 70, in render_body

I Might be able to find the line by myself, but not sure what to change about it, and the system is in use at work, so i rly don't wanna mess it up, as all other functions are working.

Thanks!

Thrakon, 2015-12-30:
Update: I removed the log part of the HAML file;

'
          - if viewlog:
            %table.table.table-striped
              - for line in viewlog:
                %tr
                  %td ${line}
          - else:
            %p There are no recent log messages.
'

Got rid of the Error.

vpetersson, 2015-12-30:
@Thrakon Looks like a unicode error. Could you take a quick look @over64?

over64, 2015-12-30:
@Thrakon, could you attach template file

Thrakon, 2015-12-30:
@over64 Hi! Thanks for looking at my case. I'll be back at work on monday, can't reach the raspberry pies from the outside. Can you specify what file you are talking about?

over64, 2016-01-04:
@Thrakon, screenly/templates/views/settings.haml

Thrakon, 2016-01-04:
@over64 Here is the settings.haml pulled from the RPI; [settings.txt](https://github.com/wireload/screenly-ose/files/77026/settings.txt)

over64, 2016-01-04:
@Thrakon, oops, some mistake. system_info.haml, sure

Thrakon, 2016-01-04:
@over64 System_Info haml is in the comment over :) Give me a heads up if you need anything else. Thanks!

over64, 2016-01-04:
@Thrakon, bug reproduced. The problem in logs. Your log file (/tmp/screenly_viewer.log) contains some not ascii symbol. For quick workaround you can clear log file. Will work with this

vpetersson, 2016-01-04:
@over64 Yeah, we've been haunted by that in the past. We should take a closer look to ensure that we got proper unicode support everywhere.

Kentzo, 2016-01-05:
And we can make codec to either 'ignore' or 'replace' instead of replying with 'error'.

Thrakon, 2016-01-06:
@vpetersson @over64 Confirmed that it was the norwegian letter ÆÅ that made the errors, renamed the assets to replace Å with aa and Ø with oe, works now. When you get an update to fix this, please tell me. Thanks

vpetersson, 2016-01-06:
@Thrakon Classic unicode error. Yeah, we need to make sure that we treat all input properly.

@Kentzo Isn't it better to simply handle it as proper UTF8 instead?

qbicdesign, 2016-01-08:
Same issue for me using Estonian characters öäüõ
Seems that the error only started appearing after running the upgrade script.
Before I upgraded, pretty sure I could visit that page no problem.

anayrat, 2016-04-19:
Any news on this issue?
Thanks

qbicdesign, 2016-04-19:
Sorry no, haven't updated yet so I don't know if its fixed.

Phil Mills
Qbic OÜ
+372 53483210
qbicdesign@gmail.com

vpetersson, 2016-05-30:
We've now landed this in the development branch. Will be merged to production in the next version.

closed 2016-05-30 by vpetersson