Screenly/Anthias #345: Slide show website displays only 2 slides - if more than 1 asset webpage is active

WTPLJim, 2015-12-23:
When a slideshow URL is provided as an asset (from SmugMug gallery) the display works fine... 
Until a second gallery slide show becomes active. Once more than one slideshow URL is provided, Screenly displays the first 2 images of the first slideshow; then moves to the next show, displays two images, and repeats after all slide shows are cycled.

Slide shows work independently, but only 1 can be active and display all photos.

WTPLJim, 2015-12-23:
link format: https://whitehallpl.smugmug.com/frame/slideshow?key=d7Qbjk&autoStart=1&captions=0&navigation=0&playButton=0&speed=3&transition=fade&transitionSpeed=2" width="800" height="600" frameborder="no" scrolling="no"

vpetersson, 2015-12-23:
Yeah, that won't work very well for a number of reasons. Instead, upload them as native assets in Screenly, and you'll have a much better result.

WTPLJim, 2015-12-23:
Can you explain why the one slide show works fine; but falls apart when more are added? thanks.

vpetersson, 2015-12-23:
Here are a few reasons:

 * The web browser is not very powerful. As such, heavy JavaScript etc won't work well.
 * It makes little sense to view a third party gallery, as you'd get a much better result with native assets.

closed 2015-12-23 by vpetersson